openwrt/packages #13148: Boost: please update to 1.74.0

wongsyrone, 2020-08-17:
Maintainer: @ClaymorePT

Description:

1.74.0 is released, please update the package.

ClaymorePT, 2020-08-17:
@wongsyrone  I'm aware of this. Thanks anyway for the heads-up.

neheb, 2020-08-18:
Curious what's so special about 1.74.

ClaymorePT, 2020-08-18:
@neheb On a first look, 1.74 does not need current patch `003-beast-coroutines.patch` for beast. Fix is already included. The others are still needed and they will need to be re-created.

It also seems that the libraries which are being more actively maintained, have introduced updates to be build with C++20.

Then, there's a bunch of fixes and improvements for most libraries. Boost.ASIO seems to contain a bunch of important fixes according to their log.

I'll be providing an update when I can.

wongsyrone, 2020-08-18:
002-outcome-assert.patch and 003-beast-coroutines.patch can be removed, BTW, I'm a heavy user of Boost.asio

ClaymorePT, 2020-08-18:
@wongsyrone  Yep `002-outcome-assert.patch` is also not needed. In truth, it was never needed. I don't really remember why it was added in the first place, since it was only adding a comment.

wongsyrone, 2020-08-18:
It's from https://www.boost.org/users/history/version_1_73_0.html, it removes an assertion as well.

ClaymorePT, 2020-08-18:
@wongsyrone ah, yes, you're right. There's two of them. Well, with this update, I'm also adding support to build Boost with C++20.

ClaymorePT, 2020-08-18:
@wongsyrone  @neheb 

-> https://github.com/openwrt/packages/pull/13160

closed 2020-08-18 by neheb