openwrt/packages #13178: unbound: allow forwarding to localhost without passing dns_assist

alaviss, 2020-08-20:
Maintainer: @EricLuehrsen 
Environment: OpenWrt master

Description:
With the introduction of `dns_assist` from https://github.com/openwrt/packages/commit/e81529596df88eae272c00bf8ac69b15e19a5bd5, unbound init script "helpfully" translated this:

```
config zone
	option enabled '1'
	option fallback '1'
	option zone_type 'forward_zone'
	list server '127.0.0.1@5053'
	list server '127.0.0.1@5054'
	list server '127.0.0.1@5055'
	list server '127.0.0.1@5056'
	list server '127.0.0.1@5057'
	list zone_name '.'
```

into:

```
forward-zone:
  name: .
  forward-host: 127.0.0.1@5053
  forward-host: 127.0.0.1@5054
  forward-host: 127.0.0.1@5055
  forward-host: 127.0.0.1@5056
  forward-host: 127.0.0.1@5057
  forward-first: yes
  forward-tls-upstream: no
```

When `dns_assist` is undefined, which is completely wrong (`forward-host` should be `forward-addr`) and breaks forwarding to `https-dns-proxy`, trashing my router DNS. An option to override this "smart" feature is very much needed.

The walkaround right now is to override the fallback `dns_ast=0` to `dns_ast=1` here:

https://github.com/openwrt/packages/blob/95362efd29cef6433c820fd15bf33e73637a336c/net/unbound/files/unbound.sh#L479-L482

EricLuehrsen, 2020-08-21:
The file `unbound_srv.conf`is included at the end of the `server:` clause, and the file `unbound_ext.conf` is included after all other clauses so you can add less common Unbound features. I would suggest using these when UCI does not completely cover your use case. Your use case is single clause and static in either format.

UCI (and LuCI) has diagnostics around 127.0.0.0/8 because it is possible to create unpleasant consequences. The implementation for UCI here is targeted at amateur or simpler use cases. Unbound is intended for professional use, and doesn't have some of the clever checking like dnsmasq. If a local host forwarding loop is created, then Unbound will happily follow it as you instructed until the router crashes. Unbound also has extensive configuration and tuning which would be infeasible to replicate in UCI.

However the rejected configuration case is not robust and should not write anything rather than invalid configuration. Also `htpps-dns-proxy` is a program that has maintenance in OpenWrt, so it can be a testable helper program.

alaviss, 2020-08-21:
> UCI (and LuCI) has diagnostics around 127.0.0.0/8 because it is possible to create unpleasant consequences. The implementation for UCI here is targeted at amateur or simpler use cases. Unbound is [...]

Yes, but I'd like to keep most of my configuration in UCI instead of spreading it out over two files. I don't think there would be too many problem with providing a switch that let users tell UCI that we know what we are doing and settings should be applied as is. I don't mind having smart features to catch errors early on, but there should always be a way to easily override them IMO.

PR #13841 (2020-11-05): unbound: update to 1.1.12
Maintainer: me
Tested: Linksys WRT3200ACM
Description: resolves #13178, resolves #13623, resolves #13750, resolves #13852
If DoH is enabled, then a few glitches in the upstream tuning options that are being worked out. Default compile option is disabled, but it is available if someone wants to try it.
closed 2020-11-05 by neheb