openwrt/packages #13196: node-homebridge-cmd4: Node modules dependency problem

zokl, 2020-08-24:
Hello,
I am not sure if the described bug is for this issue tracker but I try it :-)

I am trying to prepare my own Node.js module ([https://github.com/zokl/openwrt-node-packages/blob/master/node-homebridge-cmd4/Makefile](homebridge-cmd4)) but I found a problem with C lib dependencies. All modules that I install are pure JavaScript but after the end of the installation process, "make" fails with the following error. I tried to add +libc +libpthread to DEPENDS but without success.

```
mkdir -p /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/.pkgdir/node-homebridge-cmd4
install -d -m0755 /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/.pkgdir/node-homebridge-cmd4/usr/lib/node_modules/homebridge-cmd4
cp -fpR /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-install/usr/lib/node_modules/homebridge-cmd4/{package.json,LICENSE,*.md,*.js} /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/.pkgdir/node-homebridge-cmd4/usr/lib/node_modules/homebridge-cmd4
cp -fpR /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-install/usr/lib/node_modules/homebridge-cmd4/{node_modules,Extras} /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/.pkgdir/node-homebridge-cmd4/usr/lib/node_modules/homebridge-cmd4
touch /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/.pkgdir/node-homebridge-cmd4.installed
mkdir -p /mnt/data/homebridge_a20/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/stamp
SHELL= flock /mnt/data/homebridge_a20/tmp/.root-copy.flock -c 'cp -fpR /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/.pkgdir/node-homebridge-cmd4/. /mnt/data/homebridge_a20/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/'
touch /mnt/data/homebridge_a20/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/stamp/.node-homebridge-cmd4_installed
mkdir -p /mnt/data/homebridge_a20/bin/targets/sunxi/cortexa7/packages /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-arm_cortex-a7_neon-vfpv4/node-homebridge-cmd4/CONTROL /mnt/data/homebridge_a20/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo
install -d -m0755 /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-arm_cortex-a7_neon-vfpv4/node-homebridge-cmd4/usr/lib/node_modules/homebridge-cmd4
cp -fpR /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-install/usr/lib/node_modules/homebridge-cmd4/{package.json,LICENSE,*.md,*.js} /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-arm_cortex-a7_neon-vfpv4/node-homebridge-cmd4/usr/lib/node_modules/homebridge-cmd4
cp -fpR /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-install/usr/lib/node_modules/homebridge-cmd4/{node_modules,Extras} /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-arm_cortex-a7_neon-vfpv4/node-homebridge-cmd4/usr/lib/node_modules/homebridge-cmd4
find /mnt/data/homebridge_a20/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-homebridge-cmd4-2.3.2/ipkg-arm_cortex-a7_neon-vfpv4/node-homebridge-cmd4 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package node-homebridge-cmd4 is missing dependencies for the following libraries:
libc.so.6
libm.so.6
libpthread.so.0
make[2]: *** [Makefile:78: /mnt/data/homebridge_a20/bin/packages/arm_cortex-a7_neon-vfpv4/nodeextras/node-homebridge-cmd4_2.3.2-1_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[2]: Leaving directory '/mnt/data/homebridge_a20/feeds/nodeextras/node-homebridge-cmd4'
time: package/feeds/nodeextras/node-homebridge-cmd4/compile#30.93#9.92#42.72
make[1]: *** [package/Makefile:113: package/feeds/nodeextras/node-homebridge-cmd4/compile] Error 2
make[1]: Leaving directory '/mnt/data/homebridge_a20'
make: *** [/mnt/data/homebridge_a20/include/toplevel.mk:235: package/node-homebridge-cmd4/compile] Error 2
```

If I run make with "-i" (suppress errors) compilation process finish and modules are usable. The checking process fails only if I copy the content of the "node_modules" directory to the target system.

Is the problem in prepared Makefile or somewhere in OpenWRT package infrastructure?

Thank you,
Zbyněk Kocur

nxhack, 2020-08-26:
@zokl 
We're not talking about node.js stuff in this packages repository, so let's discuss it in a different place.

https://github.com/nxhack/openwrt-node-packages/issues/695

zokl, 2020-08-26:
Closing because of moving to another place.

closed 2020-08-26 by zokl