openwrt/packages #13199: BusyBox from 1.30.1-5 to  1.30.1-6 Not support ip tunnel add

lelguea, 2020-08-24:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:
When i update busybox from 1.30.1-5 to 1.301-6 my tunnel not work.
I need to reinstall Openwrt to use old version of busybox and use ip tunnel.

neheb, 2020-08-25:
ping @rsalvaterra

rsalvaterra, 2020-08-25:
On Tue, 25 Aug 2020 at 06:36, Rosen Penev <notifications@github.com> wrote:
>
> ping @rsalvaterra

No idea. I only ever touched ip rule (and my patch hasn't even been
applied yet), which has nothing to do with ip tunnel. And ip tunnel
itself hasn't changed in a while [1]. Are you sure this is a BusyBox
issue?

[1] https://git.busybox.net/busybox/log/networking/libiproute/iptunnel.c

lelguea, 2020-08-25:
Yes, is BusyBox issue.  With 1.30.1.5 Work greats....

[image: image.png]
MTIA
*Lorenzo Miguel Elguea Fernández*
Director de TI
Tecnologías de Información
Campus Ciudad de México
*lelguea@up.edu.mx* / Tel. 55 5482 1600 / 5500 / www.up.edu.mx
 facebook.com/UPMexico <https://www.facebook.com/UPMexico/>   @UPMexico
<https://twitter.com/UPMexico>  <https://twitter.com/UPbonaterra>  UPMexico
<https://www.instagram.com/upmexico/>  <https://www.instagram.com/upmexico/>

-- 
*POLÍTICA DE 
PRIVACIDAD: Las instituciones pertenecientes al Sistema 
UP-IPADE 
utilizarán cualquier dato personal expuesto en el presente correo 

electrónico, única y exclusivamente para cuestiones académicas, 

administrativas, de comunicación, o bien para las finalidades expresadas
 
en cada asunto en concreto, esto en cumplimiento con la Ley Federal de 

Protección de Datos Personales en Posesión de los Particulares. Para 
mayor 
información acerca del tratamiento y de los derechos que puede 
hacer 
valer, usted puede acceder al aviso de privacidad integral a 
través de 
nuestras páginas de Internet: www.up.edu.mx <http://www.up.edu.mx> / 
prepaup.up.edu.mx <http://prepaup.up.edu.mx> / www.ipade.mx 
<http://www.ipade.mx> / www.ipadealumni.com.mx 
<http://www.ipadealumni.com.mx> 
La información contenida en este correo es 
privada y confidencial, dirigida exclusivamente a su destinatario. Si usted 
no es el destinatario del mismo debe destruirlo y notificar al remitente 
absteniéndose de obtener copias, ni difundirlo por ningún sistema, ya que 
está prohibido y goza de la protección legal de las comunicaciones.*

neheb, 2020-08-25:
The change is here: https://github.com/openwrt/openwrt/commit/e754e0a143643d2d0460aec193fc20b6ca457614

I have no idea why you're having issues. The change is no-op.

There are also these ones:

https://github.com/openwrt/openwrt/commit/dc97db2e39fe59f150c8b4f913cd0ccaea4702d8
https://github.com/openwrt/openwrt/commit/cde70954effe137fac830730105853f929e9d28a
https://github.com/openwrt/openwrt/commit/55c29c398c2f4f290b5154c8daeff0f5069273b4

I don't see anything there that could break ip tunnel.

rsalvaterra, 2020-08-27:
Uh… Hold up. BusyBox's ip tunnel applet isn't built by default, I just
checked [1]. Is this a standard or a custom build?

Cheers,
Rui

[1] https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/utils/busybox/Config-defaults.in;h=29724041f47ab9cfa456149b9b8f29b5d7d92693;hb=HEAD#l2352

lelguea, 2020-08-27:
In the version 1.30.1-5 is working..

In my startup script i have:

ip tunnel add up-ipv6 mode sit remote x.x.x.x ttl 255
ip link set up-ipv6 up
ip addr add 2001:DB8::/64 dev up-ipv6
ip route add ::/0 dev up-ipv6

And in 1.30.1-6 not working....

MTIA
*Lorenzo Miguel Elguea Fernández*
Director de TI
Tecnologías de Información
Campus Ciudad de México
*lelguea@up.edu.mx* / Tel. 55 5482 1600 / 5500 / www.up.edu.mx
 facebook.com/UPMexico <https://www.facebook.com/UPMexico/>   @UPMexico
<https://twitter.com/UPMexico>  <https://twitter.com/UPbonaterra>  UPMexico
<https://www.instagram.com/upmexico/>  <https://www.instagram.com/upmexico/>

-- 
*POLÍTICA DE 
PRIVACIDAD: Las instituciones pertenecientes al Sistema 
UP-IPADE 
utilizarán cualquier dato personal expuesto en el presente correo 

electrónico, única y exclusivamente para cuestiones académicas, 

administrativas, de comunicación, o bien para las finalidades expresadas
 
en cada asunto en concreto, esto en cumplimiento con la Ley Federal de 

Protección de Datos Personales en Posesión de los Particulares. Para 
mayor 
información acerca del tratamiento y de los derechos que puede 
hacer 
valer, usted puede acceder al aviso de privacidad integral a 
través de 
nuestras páginas de Internet: www.up.edu.mx <http://www.up.edu.mx> / 
prepaup.up.edu.mx <http://prepaup.up.edu.mx> / www.ipade.mx 
<http://www.ipade.mx> / www.ipadealumni.com.mx 
<http://www.ipadealumni.com.mx> 
La información contenida en este correo es 
privada y confidencial, dirigida exclusivamente a su destinatario. Si usted 
no es el destinatario del mismo debe destruirlo y notificar al remitente 
absteniéndose de obtener copias, ni difundirlo por ningún sistema, ya que 
está prohibido y goza de la protección legal de las comunicaciones.*

neheb, 2020-09-19:
Closing since this belongs in https://bugs.openwrt.org

closed 2020-09-19 by neheb

lelguea, 2020-09-20:
This is the error in BusyBox 1.30.1.6

root@OpenWrt:/tmp# ip tunnel add he-ipv6 mode sit remote 216.xx.yy.zz local
xxx.xxx.xxx.xxx ttl 255
BusyBox v1.30.1 () multi-call binary.

Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]

MTIA
*Lorenzo Miguel Elguea Fernández*
Director de TI
Tecnologías de Información
Campus Ciudad de México
*lelguea@up.edu.mx* / Tel. 55 5482 1600 / 5500 / www.up.edu.mx
 facebook.com/UPMexico <https://www.facebook.com/UPMexico/>   @UPMexico
<https://twitter.com/UPMexico>  <https://twitter.com/UPbonaterra>  UPMexico
<https://www.instagram.com/upmexico/>  <https://www.instagram.com/upmexico/>

-- 
*POLÍTICA DE 
PRIVACIDAD: Las instituciones pertenecientes al Sistema 
UP-IPADE 
utilizarán cualquier dato personal expuesto en el presente correo 

electrónico, única y exclusivamente para cuestiones académicas, 

administrativas, de comunicación, o bien para las finalidades expresadas
 
en cada asunto en concreto, esto en cumplimiento con la Ley Federal de 

Protección de Datos Personales en Posesión de los Particulares. Para 
mayor 
información acerca del tratamiento y de los derechos que puede 
hacer 
valer, usted puede acceder al aviso de privacidad integral a 
través de 
nuestras páginas de Internet: www.up.edu.mx <http://www.up.edu.mx> / 
prepaup.up.edu.mx <http://prepaup.up.edu.mx> / www.ipade.mx 
<http://www.ipade.mx> / www.ipadealumni.com.mx 
<http://www.ipadealumni.com.mx> 
La información contenida en este correo es 
privada y confidencial, dirigida exclusivamente a su destinatario. Si usted 
no es el destinatario del mismo debe destruirlo y notificar al remitente 
absteniéndose de obtener copias, ni difundirlo por ningún sistema, ya que 
está prohibido y goza de la protección legal de las comunicaciones.*