openwrt/packages #13210: libgd: libjpeg dependency broken

jonozzz, 2020-08-25:
```
# /usr/bin/vnstati -t -i eth0 -o -
Error loading shared library libjpeg.so.9: No such file or directory (needed by /usr/lib/libgd.so.3)
Error relocating /usr/lib/libgd.so.3: jpeg_save_markers: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_set_quality: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_finish_compress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_destroy: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_write_scanlines: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_set_defaults: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_destroy_compress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_start_decompress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_CreateDecompress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_write_marker: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_std_error: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_read_header: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_start_compress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_CreateCompress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_finish_decompress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_simple_progression: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_read_scanlines: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_destroy_decompress: symbol not found
Error relocating /usr/lib/libgd.so.3: jpeg_resync_to_restart: symbol not found
```

To workaround I manually symlinked that missing .so:
`ln -s /usr/lib/libjpeg.so /usr/lib/libjpeg.so.9`

@neheb I think this started happening once libjpeg was removed

neheb, 2020-08-25:
It looks like your staging_dir is dirty. Run make clean and recompile.

jonozzz, 2020-08-25:
yup, thanks.

closed 2020-08-25 by jonozzz