openwrt/packages #13240: The TTYD under the master branch cannot be connected after updating！！

zxlhhyccc, 2020-08-28:
It turns out that the library connection is wrong: it depends on libwebsockets. So.14 but libwebsockets. So.15, because the compiled libwebsocketsy source package is actually libwebsockets. So.14.

[image]

[image]

neheb, 2020-08-28:
Sounds like a dirty staging_dir

zxlhhyccc, 2020-08-28:
> Sounds like a dirty staging_dir

This problem still exists with the newly cleaned original compiled firmware!

1715173329, 2020-08-28:
> Remove the newly added patch

Sure. One the source is changed, openwrt will re-compile it automatically.

The reason why you linked to wrong library is someone update the source, but you didn't recompile it.

1715173329, 2020-08-28:
> Sounds like a dirty staging_dir

Actually yes, simply running `make package/ttyd/{clean,compile}`  and the issue will be solved.

But with existing compilation mechanism, if the dependency of a package was updated, the package won't recompile automatically, and this caused some issues.
Maybe we need a improvement.

zxlhhyccc, 2020-08-29:
> > Sounds like a dirty staging_dir
> 
> Not yet, simply running `make package/ttyd/{clean,compile}` and the issue will be solved.
> 
> With existing compilation mechanism, if the dependencies of a package were updated, the package won't be recompiled automatically, and this caused some issues.
> Maybe we need a improvement.

Let me try to clean all the dependent libraries of TTYD and TTYD itself and recompile them. I just don't understand. I am compiled by the cloud, and there is also the problem that I cannot connect？

zxlhhyccc, 2020-08-29:
> Not yet, simply running `make package/ttyd/{clean,compile}` and the issue will be solved.
>
> With existing compilation mechanism, if the dependencies of a package were updated, the package won't be [...]

I recompile lean and recompile all the dependent libraries of TTYD and TTYD and connected normally, but I just don't understand. The cloud compiled is a brand new compilation. Why did the connection fail?

closed 2020-08-29 by zxlhhyccc