openwrt/packages #13242: Failure due to SIGSEGV in nlbwmon

n7ckn, 2020-08-29:
According to my system logs, nlbwmon no longer works due to:

Sat Aug 29 08:58:08 2020 kern.info kernel: [39011.829126] do_page_fault(): sending SIGSEGV to nlbw for invalid write access to 00000000
Sat Aug 29 08:58:08 2020 kern.info kernel: [39011.829146] epc = 77e810e4 in libc.so[77e64000+98000]
Sat Aug 29 08:58:08 2020 kern.info kernel: [39011.829199] ra  = 00404311 in nlbwmon[400000+6000]
Sat Aug 29 08:58:08 2020 kern.info kernel: [39012.539976] do_page_fault(): sending SIGSEGV to nlbw for invalid write access to 00000000
Sat Aug 29 08:58:08 2020 kern.info kernel: [39012.539994] epc = 77ecd0e4 in libc.so[77eb0000+98000]
Sat Aug 29 08:58:08 2020 kern.info kernel: [39012.540032] ra  = 00404311 in nlbwmon[400000+6000]

This is logged as soon as I try to open the luci app.

I have removed and reinstalled nlbwmon and the luci app and tried reflashing the firmware and the issue still persists.

neheb, 2020-08-30:
Try using strace on nlbwmon.

n7ckn, 2020-08-30:
strace output:

execve("/usr/sbin/nlbwmon", ["nlbwmon"], 0x7fcfc570 /* 12 vars */) = 0
set_thread_area(0x77f47efc)             = 0
set_tid_address(0x77f3f008)             = 6651
open("/etc/ld-musl-mips-sf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libnl-tiny.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libnl-tiny.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libnl-tiny.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=25456, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\32\320\0\0\0004"..., 936) = 936
mmap2(NULL, 94208, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77e7e000
mmap2(0x77e93000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x5000) = 0x77e93000
close(3)                                = 0
open("/lib/libubox.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=33344, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\36P\0\0\0004"..., 936) = 936
mmap2(NULL, 102400, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77e64000
mmap2(0x77e7b000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x7000) = 0x77e7b000
close(3)                                = 0
open("/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libz.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=61720, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\27\0\0\0\0004"..., 936) = 936
mmap2(NULL, 131072, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77e44000
mmap2(0x77e62000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xe000) = 0x77e62000
close(3)                                = 0
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=78080, ...}) = 0
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0'\320\0\0\0004"..., 936) = 936
mmap2(NULL, 147456, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x77e20000
mmap2(0x77e42000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x12000) = 0x77e42000
close(3)                                = 0
mprotect(0x77e93000, 4096, PROT_READ)   = 0
mprotect(0x77e7b000, 4096, PROT_READ)   = 0
mprotect(0x77e62000, 4096, PROT_READ)   = 0
mprotect(0x77e42000, 4096, PROT_READ)   = 0
mprotect(0x415000, 4096, PROT_READ)     = 0
getpid()                                = 6651
socket(AF_NETLINK, SOCK_RAW, NETLINK_ROUTE) = 3
setsockopt(3, SOL_SOCKET, SO_SNDBUF, [32768], 4) = 0
setsockopt(3, SOL_SOCKET, SO_RCVBUF, [32768], 4) = 0
bind(3, {sa_family=AF_NETLINK, nl_pid=6651, nl_groups=00000000}, 12) = 0
getsockname(3, {sa_family=AF_NETLINK, nl_pid=6651, nl_groups=00000000}, [12]) = 0
writev(2, [{iov_base="", iov_len=0}, {iov_base="No interval specified; assuming "..., iov_len=45}], 2No interval specified; assuming 1st of month
) = 45
epoll_create1(0)                        = 4
fcntl64(4, F_GETFD)                     = 0
fcntl64(4, F_SETFD, FD_CLOEXEC)         = 0
pipe([5, 6])                            = 5
fcntl64(5, F_GETFD)                     = 0
fcntl64(5, F_SETFD, FD_CLOEXEC)         = 0
fcntl64(5, F_GETFL)                     = 0 (flags O_RDONLY)
fcntl64(5, F_SETFL, O_RDONLY|O_NONBLOCK|O_LARGEFILE) = 0
fcntl64(6, F_GETFD)                     = 0
fcntl64(6, F_SETFD, FD_CLOEXEC)         = 0
fcntl64(6, F_GETFL)                     = 0x1 (flags O_WRONLY)
fcntl64(6, F_SETFL, O_WRONLY|O_NONBLOCK|O_LARGEFILE) = 0
fcntl64(5, F_GETFL)                     = 0x80 (flags O_RDONLY|O_NONBLOCK)
fcntl64(5, F_SETFL, O_RDONLY|O_NONBLOCK|O_LARGEFILE) = 0
epoll_ctl(4, EPOLL_CTL_ADD, 5, {EPOLLIN|EPOLLRDHUP, {u32=2011676688, u64=8640085585085595648}}) = 0
rt_sigaction(SIGINT, NULL, {sa_handler=SIG_DFL, sa_mask=[RT_68 RT_70 RT_71 RT_72 RT_73 RT_75 RT_77 RT_78 RT_79 RT_80 RT_81 RT_82 RT_85 RT_86 RT_87 RT_88 RT_89 RT_90 RT_91 RT_93 RT_94 RT_95], sa_flags=0}, 16) = 0
rt_sigprocmask(SIG_UNBLOCK, [RT_1 RT_2], NULL, 16) = 0
rt_sigaction(SIGINT, {sa_handler=0x77e67975, sa_mask=[RT_68 RT_70 RT_71 RT_73 RT_74 RT_75 RT_77 RT_78 RT_82 RT_84 RT_86 RT_87 RT_88 RT_89 RT_90 RT_91 RT_93 RT_94 RT_95], sa_flags=SA_RESTORER, sa_restorer=NULL}, NULL, 16) = 0
rt_sigaction(SIGTERM, NULL, {sa_handler=SIG_DFL, sa_mask=[RT_68 RT_70 RT_71 RT_72 RT_73 RT_75 RT_77 RT_78 RT_79 RT_80 RT_81 RT_82 RT_85 RT_86 RT_87 RT_88 RT_89 RT_90 RT_91 RT_93 RT_94 RT_95], sa_flags=0}, 16) = 0
[... 9 lines cut ...]
brk(NULL)                               = 0xb34000
brk(0xb37000)                           = 0xb37000
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 7
fstat64(7, {st_mode=S_IFREG|0644, st_size=25, ...}) = 0
mmap2(NULL, 25, PROT_READ, MAP_SHARED, 7, 0) = 0x77f3c000
close(7)                                = 0
stat64("/tmp/0.db.gz", 0x7fdb9280)      = -1 ENOENT (No such file or directory)
stat64("/tmp/0.db", 0x7fdb9280)         = -1 ENOENT (No such file or directory)
munmap(0x77f3c000, 25)                  = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 7
fstat64(7, {st_mode=S_IFREG|0644, st_size=25, ...}) = 0
[... cut ...]
open("/usr/share/nlbwmon/protocols", O_RDONLY|O_LARGEFILE) = 7
read(7, "0 0 IPv6-HOPOPT(0)\n1 0 ICMP(0)\n2"..., 1024) = 1024
read(7, "boots(1067)U\n6 1080 socks(1080)T"..., 1024) = 1024
read(7, "S(5353)U\n17 5355 llmnr(5355)U\n6 "..., 1024) = 1024
read(7, "9153 airplay(49153)T\n6 49154 air"..., 1024) = 432
read(7, "", 1024)                       = 0
close(7)                                = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault

closed 2020-09-03 by n7ckn