openwrt/packages #13245: imagebuilder : “following packages conflict with busybox”

sinancetinkaya, 2020-08-29:
Supply the following if possible:
- Device problem occurs on
unielec_u7621-06-16m

- Software versions of OpenWrt/LEDE release, packages, etc.
latest snapshots

- Steps to reproduce
```
sinan@sinan-VirtualBox:~$ make image PROFILE=unielec_u7621-06-16m PACKAGES="luci block-mount kmod-fs-ext4 wpad-mini nano htop kmod-mt76 kmod-mt76-core kmod-mt76x2 luci-app-sqm sqm-scripts kmod-usb-storage-uas kmod-usb3"
......................
......................
......................
Collected errors:
 * check_conflicts_for: The following packages conflict with busybox:
 * check_conflicts_for: 	busybox-selinux * 
 * opkg_install_cmd: Cannot install package busybox.
Makefile:154: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:112: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:207: recipe for target 'image' failed
make: *** [image] Error 2
```

coolermister, 2020-09-02:
+1
Reproduced it too, even when building from sources. If you add some package that provide the same tool which already included in busybox, for example ifenslave, ifup/ifdown, and so on, build says "trying to install smth, but that provided by package busybox".
I used to think that it's normal behaviour)

sinancetinkaya, 2020-09-18:
Builds still have it. What's the root cause of this problem?

dangowrt, 2020-09-18:
Apparently what happens is that `procd-selinux` is somehow preferred over `procd` as it `PROVIDES:=procd`.
Can you try building an ImageBuilder from source and try adding `PROVIDES:=procd` in package/system/procd/Makefile also to the procd package? Same for busybox I suppose.

sinancetinkaya, 2020-09-18:
@dangowrt 

> try building an ImageBuilder from source
How? My linux knowledge is very limited.

aparcar, 2020-09-18:
@sinancetinkaya I'll give it a test :)

aparcar, 2020-09-19:
Have you tried other targets? I'm trying to reproduce it on x86/64 and I don't see a problem

sinancetinkaya, 2020-09-19:
> Have you tried other targets? I'm trying to reproduce it on x86/64 and I don't see a problem

I wrote the build command I used in the first post. Can you try the exact command?

neheb, 2020-09-19:
Closing as this belongs in https://bugs.openwrt.org

closed 2020-09-19 by neheb