openwrt/packages #13255: Samba4 error , the reading speed is very slow

9000000, 2020-08-31:
OpenWrt SNAPSHOT r14311-e5e54e52f7
Xiaomi r3p  MT7621

```
Mon Aug 31 12:49:43 2020 daemon.err smbd[10176]: [2020/08/31 12:49:43.578449,  0] ../../lib/param/loadparm.c:801(lpcfg_map_parameter)
Mon Aug 31 12:49:43 2020 daemon.err smbd[10176]:   Unknown parameter encountered: "write cache size"
Mon Aug 31 12:49:43 2020 daemon.err smbd[10176]: [2020/08/31 12:49:43.579245,  0] ../../lib/param/loadparm.c:1841(lpcfg_do_global_parameter)
Mon Aug 31 12:49:43 2020 daemon.err smbd[10176]:   Ignoring unknown parameter "write cache size"

```

neheb, 2020-08-31:
ping @Andy2244

Andy2244, 2020-08-31:
Not an error, the samba logger uses `daemon.err` for all types of log levels, no clue why.
`write cache size` is unknown because the experimental option `ENABLE_EXTRA_TUNING` is used with snapshots 4.12.x, where this option was removed, so thats just a info/warning.

closed 2020-09-01 by 9000000