openwrt/packages #13265: Dependency on 'libcurl', which does not exist

coolermister, 2020-08-31:
Maintainer: ???
Environment: 
Target arch - x86_64
OpenWRT version - master
Build os: ubuntu server 18.04
GCC: doesn't matter 9.x or 10.x
Description:
```
WARNING: Makefile 'package/feeds/packages/apache/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/asterisk-16.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/boinc/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/czmq/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/domoticz/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/elektra/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/esniper/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/forked-daapd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/freeradius3/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/gammu/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/git/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/gnupg/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/hcxtools/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/hs20/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/https-dns-proxy/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/icecast/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/idevicerestore/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ipfs-http-client/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ipfs-http-client/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/leech/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/libmariadb/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/libnpupnp/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/libulfius/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/libulfius/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/linknx/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/luarocks/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/measurement-kit/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/measurement-kit/Makefile' has a build dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/mocp/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/mpd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/mpd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/mpd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/netifyd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/netopeer2/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/noddos/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ooniprobe/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/pdns/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/pdns/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/perl-www-curl/Makefile' has a dependency on 'libcurl', which does not exist
[... 4 lines cut ...]
WARNING: Makefile 'package/feeds/packages/rtorrent/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/strongswan/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/syslog-ng/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/transmission/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/transmission/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/transmission/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/transmission/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/transmission/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/transmission/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/uacme/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/xmlrpc-c/Makefile' has a dependency on 'libcurl', which does not exist
[... cut ...]
WARNING: Makefile 'package/feeds/packages/ykclient/Makefile' has a build dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ykpers/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/yubico-pam/Makefile' has a dependency on 'curl', which does not exist
```
and all those packages won't build(

Steps to reproduce:
1. ./scrips/feeda update -a && ./scrips/feeda install -a
2. make menuconfig
3. ???
4. profit

This may be invented by [this merge?](https://github.com/openwrt/packages/pull/13214 )

neheb, 2020-08-31:
Use feeds install -a -f

coolermister, 2020-09-02:
Thank's a lot! That helped)

closed 2020-09-02 by coolermister

ahmar20, 2020-09-03:
> Use feeds install -a -f

This should be the default option to install packages in the source. I was having the same issue for 2 days, I thought it was an issue with the build system.