openwrt/packages #13270: alsa-lib: missing files

Vladdrako, 2020-09-01:
USB sound card not working due to errors:
```
root@MI-R3G:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 1b3f:2008 Generalplus Technology Inc.
Bus 001 Device 006: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@MI-R3G:~# lsusb -v -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
    |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 8, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 3: Dev 8, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 3: Dev 8, If 2, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 3: Dev 8, If 3, Class=Human Interface Device, Driver=, 12M
root@MI-R3G:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@MI-R3G:~# alsactl init
alsa-lib parser.c:2101:(parse_toplevel_config) Syntax field not found in /usr/share/alsa/ucm2/ucm.conf
alsa-lib main.c:983:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -22
Found hardware: "USB-Audio" "USB Mixer" "USB1b3f:2008" "" ""
Hardware is initialized using a generic method
root@MI-R3G:~# mpc play
RetroFM: NAVI - ZAKOHUUS`
[paused]  #1/1   0:00/0:00 (0%)
volume: n/a   repeat: off   random: off   single: off   consume: off
ERROR: Failed to open "My ALSA Device" (alsa); Error opening ALSA device "hw:0,0"; snd_pcm_hw_params() failed: I/O error
```
There is no /usr/share/alsa/ucm2 folder.

OpenWrt SNAPSHOT r14353-4841ac042e / LuCI Master git-20.244.48572-a093971
mt7621 (Xiaomi Mi Router 3G) Kernel version: 5.4.61
Maintainer: @tripolar

Vladdrako, 2020-09-01:
I have no ideas if this is alsa-lib bug or not, so sorry if I mistaked.

neheb, 2020-09-01:
Related: https://bugs.archlinux.org/task/67027

Seems to be a different package entirely. https://git.alsa-project.org/?p=alsa-ucm-conf.git . I guess it needs to be added.

closed 2020-09-23 by neheb