openwrt/packages #13275: nginx-all-module: build falls with missing dependencie libpcre.so.1

coolermister, 2020-09-02:
Maintainer: @heil @Ansuel 
Environment: 
Target arch - x86_64
OpenWRT version - master, lib musl
Build os: ubuntu server 18.04
GCC: 10.x
Description:
It looks like the other packages that use libpcre are builded correctly
Steps to reproduce:
1. ./scrips/feeda update -a && ./scrips/feeda install -a -f
2. In menuconfig select to * nginx-all-module, nginx-ssl-util, nginx-mod-luci
3. make V=s
4. Profit

This is a part of build log:
```
install -m0600 ./files/_redirect2ssl.conf /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/.pkgdir/nginx-all-module/etc/nginx/conf.d/
touch /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/.pkgdir/nginx-all-module.installed
mkdir -p /media/openwrt/bin/targets/x86/64/packages /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/CONTROL /media/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/usr/sbin
install -m0755 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-install/usr/sbin/nginx /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/usr/sbin/
install -d -m0755 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx/conf.d
install -m0644 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-install/etc/nginx/mime.types /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-install/etc/nginx/koi-utf /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-install/etc/nginx/koi-win /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-install/etc/nginx/win-utf /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-install/etc/nginx/fastcgi_params /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-install/etc/nginx/uwsgi_params /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx/
install -m0600 ./files/nginx.conf /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx/
install -m0600 ./files/_lan.conf /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx/conf.d/
install -d -m0755 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/init.d
install -m0755 ./files/nginx.init /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/init.d/nginx
install -d -m0755 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx
install -m0755 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/nginx-naxsi/naxsi_config/naxsi_core.rules /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx
chmod 0640 /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx/naxsi_core.rules
install -m0600 ./files/_redirect2ssl.conf /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx/etc/nginx/conf.d/
find /media/openwrt/build_dir/target-x86_64_musl/nginx-all-module/nginx-1.19.2/ipkg-x86_64/nginx -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package nginx is missing dependencies for the following libraries:
libpcre.so.1
Makefile:561: recipe for target '/media/openwrt/bin/packages/x86_64/packages/nginx_1.19.2-1_x86_64.ipk' failed
make[3]: *** [/media/openwrt/bin/packages/x86_64/packages/nginx_1.19.2-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/media/openwrt/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/all-module/compile#87.25#11.22#95.05
package/Makefile:111: recipe for target 'package/feeds/packages/nginx/compile' failed
make[2]: *** [package/feeds/packages/nginx/compile] Error 2
make[2]: Leaving directory '/media/openwrt'
package/Makefile:107: recipe for target '/media/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/media/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/openwrt'
/media/openwrt/include/toplevel.mk:233: recipe for target 'world' failed
make: *** [world] Error 2

```

coolermister, 2020-09-03:
Same with all nginx packages

coolermister, 2020-09-03:
ping @heil @Ansuel @neheb guys? I'm sorry for disturbing you, but I have been suffering for a  week comparing makefiles in search of an error, and I have found nothing( The openwrt buildroot for now is too difficult for me(

neheb, 2020-09-03:
Run make clean.

coolermister, 2020-09-03:
Did the make clean, and make dirclean, same error(

coolermister, 2020-09-07:
Initialized a fresh repository, copied the .config and kernel config-5.4 into it. Still nothing

Ansuel, 2020-09-07:
I clean/compiled the nginx-ssl package multiple times but i still can't reproduce this

coolermister, 2020-09-07:
wow. @Ansuel did you try nginx-all-modules? can i provide additional information? may be config?

Ansuel, 2020-09-07:
Well nice... with all variant i reproduced the problem... let me check

Ansuel, 2020-09-07:
@coolermister i should have fixed the problem... pushing a fix

PR #13326 (2020-09-07): nginx: fix broken all-module variant
The creation of the dummy package nginx creates some problem with dependency detection for the all-module variant. Reorganize the dependency and compile nginx before the the sub-variant.

Fixes #13275

Think the ci fail as i don't have the scripts in my repo
commit 310d6c96 (2020-09-07): nginx: fix broken all-module variant

The creation of the dummy package nginx creates some problem with dependency detection for the all-module variant. Reorganize the dependency and compile nginx before the the sub-variant.

Fixes #13275
closed 2020-09-07 by hnyman