openwrt/packages #13278: mini-snmpd: update to current upstream release

arigit, 2020-09-02:
mini-snmpd: the current shipped version is 1.4; upstream is on release 1.6 - please update the package

Maintainer: @jefferyto @mtelka 

Description:

```
The currently shipped version is very old. The current version 1.6 fixes long-standing issues and adds support for 64-bit counters which has been requested for a long time.
This package is the only option for snmp support in low-memory devices where net-snmpd does not fit, please update the release when possible

```

jefferyto, 2020-09-07:
The [actual maintainer](https://github.com/openwrt/packages/blob/4ecaaad0af29b3e6a4d47485e9b4520ce55d7e0f/net/mini_snmpd/Makefile#L13) is @marcin1j.

neheb, 2020-09-19:
Merged.

closed 2020-09-19 by neheb