openwrt/packages #13301: [vpn-policy-routing] Issue: Wrong service gateway selection after restarting the router

misiek735, 2020-09-04:
Hi!

Router: Xiaomi Router 3g,
Firmware OpenWrt 19.07-SNAPSHOT r11144-2ca5a386ee / LuCI openwrt-19.07 branch git-20.197.49655-210b489
Kernel 4.14.187.

My config is based on the instruction from readme "If the Wireguard tunnel on your router is used as default routing(...)".

After restarting the router, those devices which has been configured to work through vpn does not have the access to the internet, but those devices which are configured to go through wan has normal internet access. 
When I click manually in luci the button (service control) to restart the vpn policy routing, then the service gateway is recognized properly (as vpn, not as wan). Changing the Boot Time-out from the default 30 seconds to 60 seconds does not solve the problem. 

Thank you in advance for any suggestions.

neheb, 2020-09-19:
ping @stangri

stangri, 2020-09-20:
@misiek735 like I asked in the issue you've created in my repo, please provide all the information mentioned in the HELP section of the README.

stangri, 2021-04-20:
@neheb can you please close this due to lack of information from submitter?

closed 2021-04-20 by neheb