openwrt/packages #13304: NETDATA: Unknown package 'netdata' error on RPi 4 snapshot -- maybe a build bot error

mohitgupta1918, 2020-09-04:
Maintainer: @BKPepe @diizzyy 
Environment: aarch64_cortex-a72 (RPi 4 2 GB Model)
Firmware Version	OpenWrt SNAPSHOT r14377-7a5e4f5f00 / LuCI Master git-20.246.48839-82d4384

Description: I am trying to install netdata but i am getting an error **Unknown package 'netdata'**
I tried upgrading to the latest snapshot available at the time, still getting the same error.

I think the build bot is throwing an error while compiling (assumption based on the following forum [post](https://forum.openwrt.org/t/unable-to-install-luci-using-opkg/56609/2))

https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a72/packages/netdata/compile.txt

configure: error: libuuid required but not found. Try installing 'uuid-dev' or 'libuuid-devel'.
Makefile:96: recipe for target '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a72_musl/netdata-1.23.2/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a72_musl/netdata-1.23.2/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
time: package/feeds/packages/netdata/compile#11.46#1.71#19.82 

Can someone please have a look

diizzyy, 2020-09-06:
Seems to be related to buildbot, have you tried compiling yourself?

BKPepe, 2020-09-07:
Hm. There is something happening with buildbot for aarch64_cortex-a72.
It affects many packages including netdata:
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a72/base/e2fsprogs/compile.txt and so on.

BKPepe, 2020-09-12:
And the package is there. Closing.

closed 2020-09-12 by BKPepe