openwrt/packages #13314: nginx crash after ubus module update

ptpt52, 2020-09-06:
```
[ 1281.619764] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1281.628367] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1281.633840] ra  = 00464b4b in nginx[400000+a0000]
[ 1281.651696] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1281.660358] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1281.665848] ra  = 00464b4b in nginx[400000+a0000]
[ 1283.540151] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1283.548754] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1283.554223] ra  = 00464b4b in nginx[400000+a0000]
[ 1283.569502] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1283.578286] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1283.583776] ra  = 00464b4b in nginx[400000+a0000]
[ 1286.446828] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1286.455563] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1286.461054] ra  = 00464b4b in nginx[400000+a0000]
[ 1286.476639] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1286.485303] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1286.490797] ra  = 00464b4b in nginx[400000+a0000]
[ 1290.348857] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1290.357460] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1290.362944] ra  = 00464b4b in nginx[400000+a0000]
[ 1290.378528] do_page_fault(): sending SIGSEGV to nginx for invalid read access from 00000004
[ 1290.387147] epc = 77d7e935 in libubox.so[77d7c000+17000]
[ 1290.392751] ra  = 00464b4b in nginx[400000+a0000]
```

run test on mipsel ramips

it is related to this commit: https://github.com/openwrt/packages/commit/6c3e2e927f2bd71e5dd784eddd67d80465aa5c55

without this commit, run test is fine

ping @Ansuel

Ansuel, 2020-09-06:
@ptpt52 i can't test currently but i should have found the bug... Do you have a buildroot to test the new version ?

ptpt52, 2020-09-06:
@Ansuel sure

Ansuel, 2020-09-06:
@ptpt52 then change in the nginx makefile the version for the ubus submodule to this commit and clean/compile... tell me if the problem is fixed

b2d7260dcb428b2fb65540edb28d7538602b4a26

ptpt52, 2020-09-07:
@Ansuel 
build and test, and no crash now.

PR #13323 (2020-09-07): nginx: update ubus module to fix SIGSEGV
- Bump relase
- Bump nginx_ubus_module version

Fixes #13314

Reported-by: Chen Minqiang <ptpt52@gmail.com>
commit 7ca2238f (2020-09-07): nginx: update ubus module to fix SIGSEGV

- Bump relase
- Bump nginx_ubus_module version

Fixes #13314

Reported-by: Chen Minqiang <ptpt52@gmail.com>
closed 2020-09-07 by hnyman