openwrt/packages #13318: PHP7: Timezone TZ broken

reinerotto, 2020-09-07:
Using PHP7 on openwrt. Timezone 'UTC' works perfectly, However, setting another one in /etc/config/system, i.g.
option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
option zonename 'Europe/Berlin'
breaks the following demo:

<?php echo date("Y"); echo date("H:i:s"); ?>
Output:
Fatal error: date(): Timezone database is corrupt - this should never happen! in /www/ui/test.php on line 2

The interesting fact is, that this happens both using following option in make menuconfig - languages - PHP:

*  Use system timezone data instead of php's built-in database
Use system timezone data instead of php's built-in database
Although in Makefile for PHP7 following comment:
config PHP7_SYSTEMTZDATA
bool "Use system timezone data instead of php's built-in database"
depends on PACKAGE_php7-cli || PACKAGE_php7-cgi
default y
help
Enabling this feature automatically selects the zoneinfo-core package which contains data for UTC timezone. To use other timezones you have to install the corresponding zoneinfo-... package(s).

However, there is no (other) zoneinfo-package available.

coolermister, 2020-09-07:
Did you use ./scripts/feeds install -a -f ? 
I see this 
[image]

coolermister, 2020-09-07:
Did you try ./scripts/feeds install -a -f ?
I see all of zoneinfo packages
[image]

closed 2020-09-07 by reinerotto