openwrt/packages #13322: Continuous building

coolermister, 2020-09-07:
Guys, maintainers. If you interested in continuous building, for example for most common arch as x86 or atheros and so on, i can provide one of my own servers. For example it can be 2x opteron 6172 + 64 gb ram and ssd intel 1tb. If we always will automatically build master for example, we will be see what have been gone wrong quicker

neheb, 2020-09-19:
I would talk to @jow- about this. He doesn't really read his GitHub emails, so it's best to catch him on IRC.

Currently on https://downloads.openwrt.org , packages are rebuilt mostly every day. I have no idea what the underlying hardware is.

hnyman, 2023-01-14:
Not really a packages issue.
Currently there is automatic CI PR (and in any case, the buildbot has always built packages continuously)

closed 2023-01-14 by hnyman