openwrt/packages #13325: acme: run-acme fails and corrupts uhttpd config when uhttpd also listens on ipv6 port 80

jvasileff, 2020-09-07:
Maintainer: @tohojo
Environment: N/A, but raspberry pi 4, latest from GitHub

Description:

When `uhttpd` is configured to listen on both `0.0.0.0:80` and `[::]:80` (both ipv4 and ipv6), `run-acme.sh` clears both listen directives from `/etc/config/uhttpd` and fails.

Logs:

It can be seen in the logs that `/usr/lib/acme/run-acme`'s pre checks are incorrectly run twice:

```
Mon Sep  7 11:09:26 2020 daemon.info acme: Running pre checks for openwrt.mydomain.example.com.
Mon Sep  7 11:09:26 2020 daemon.err run-acme[6866]: acme: Running pre checks for openwrt.mydomain.example.com.
Mon Sep  7 11:09:26 2020 daemon.debug acme: port80 listens: 1442/uhttpd 1442/uhttpd
Mon Sep  7 11:09:26 2020 daemon.err run-acme[6866]: acme: port80 listens: 1442/uhttpd 1442/uhttpd
Mon Sep  7 11:09:26 2020 daemon.debug acme: Found uhttpd listening on port 80; trying to disable.
Mon Sep  7 11:09:26 2020 daemon.err run-acme[6866]: acme: Found uhttpd listening on port 80; trying to disable.
Mon Sep  7 11:09:26 2020 daemon.debug acme: Found uhttpd listening on port 80; trying to disable.
Mon Sep  7 11:09:26 2020 daemon.err run-acme[6866]: acme: Found uhttpd listening on port 80; trying to disable.
Mon Sep  7 11:09:26 2020 daemon.err run-acme[6866]: uci: Entry not found
Mon Sep  7 11:09:26 2020 daemon.err acme: openwrt.mydomain.example.com: Unable to find uhttpd listen config.
Mon Sep  7 11:09:26 2020 daemon.err run-acme[6866]: acme: openwrt.mydomain.example.com: Unable to find uhttpd listen config.
Mon Sep  7 11:09:26 2020 daemon.err acme: Manually disable uhttpd or set webroot to continue.
Mon Sep  7 11:09:26 2020 daemon.err run-acme[6866]: acme: Manually disable uhttpd or set webroot to continue.

```

Code References:

The problem occurs when [pre_checks()](https://github.com/openwrt/packages/blob/89213c1a4598f3e3fe3e140ecc8207dbaca4344a/net/acme/files/run.sh#L74) iterates processes listening on port 80. On the *first* pass, line 90's `uci set uhttpd.main.listen_http=''` clears both listen directives (ipv4 and ipv6). On the *second* pass, line 84 `if [ -z "$UHTTPD_LISTEN_HTTP" ]; then` is `true`, causing the error `Unable to find uhttpd listen config`.

The reason for the second pass is that [`get_listeners`](https://github.com/openwrt/packages/blob/89213c1a4598f3e3fe3e140ecc8207dbaca4344a/net/acme/files/run.sh#L49) finds PIDs using `netstat`, and if uhttpd is configured for both ipv4 and ipv6, there will be two entries with the same PID. For example, on my system:

```
root@OpenWrt:~# netstat -nap | grep :80
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      10949/uhttpd
tcp        0      0 :::80                   :::*                    LISTEN      10949/uhttpd
```

jvasileff, 2020-09-07:
Note that the end-user workaround is to:

```
uci set uhttpd.main.listen_http=0.0.0.0:80
uci commit uhttpd.main.listen_http
```

(followed by `/etc/init.d/uhttpd restart`)

which will replace the following in `/etc/config/uhttpd`

```
config uhttpd 'main'
	list listen_http '0.0.0.0:80'
	list listen_http '[::]:80'
```

with 

```
config uhttpd 'main'
	list listen_http '0.0.0.0:80'
```

tohojo, 2020-09-08:
Ah, that won't do obviously - thanks for reporting!

Could you please check if https://github.com/tohojo/openwrt-packages/commit/bfd7312625475e37c5e739361bf9b268c88a45a1 fixes this for you?

jvasileff, 2020-09-08:
Thanks, that works! Here's the log:

```
Tue Sep  8 12:06:11 2020 daemon.debug acme: Found uhttpd listening on port 80; trying to disable.
Tue Sep  8 12:06:11 2020 daemon.err run-acme[4665]: acme: Found uhttpd listening on port 80; trying to disable.
Tue Sep  8 12:06:11 2020 daemon.debug acme: Already handled uhttpd; skipping
Tue Sep  8 12:06:11 2020 daemon.err run-acme[4665]: acme: Already handled uhttpd; skipping
```

One nit is that the save/restore process, which is basically:

```
uci set uhttpd.main.listen_http="$(uci get uhttpd.main.listen_http)"
uci commit uhttpd.main.listen_http
```

changes the config file from

```
config uhttpd 'main'
        list listen_http '0.0.0.0:80'
        list listen_http '[::]:80'
```

to

```
config uhttpd 'main'
	option listen_http '0.0.0.0:80 [::]:80'
```

which affects the way it is displayed in the luci web ui. But it's not a huge deal, the compressed format works fine.

tohojo, 2020-09-08:
Yeah, realise that, but not sure there's any way to avoid it?

jvasileff, 2020-09-08:
Yeah I figured it wouldn't be super straight forward. I'm new to OpenWrt, so I'm not aware of all of the intricacies of `uci`.

commit e6c1676c (2020-09-09): acme: Make sure we don't handle the same listener twice

If a daemon listens on multiple addresses at once, it'll show up multiple
times in get_listeners() which will clobber the config for uhttpd. Fix this
by skipping subsequent handlings of the same daemon binary.

Fixes #13325.
closed 2020-09-09 by tohojo