openwrt/packages #13361: libs/getdns: getdns_query is statically linked against libgetdns.so

izstas, 2020-09-11:
Maintainer: @jonathanunderwood
Environment: ath79, TP-Link Archer C7 v4, OpenWrt 19.07.4

Description:

Upon updating from 19.07.3 (with no packages updated since the release date) to 19.07.4, I noticed that the size of /usr/sbin/getdns_query binary has significantly increased (from about 39 KB to about 306 KB)

I found that /usr/sbin/getdns_query doesn't depend on libgetdns.so like it did in the previous version, so I guess it's linked statically. I believe this might not be an intentional change.

net/stubby does depend on libgetdns.so, as expected.

neheb, 2020-09-18:
Sounds like a bug with CMake.

jonathanunderwood, 2021-02-07:
Is this still an issue?

izstas, 2021-02-07:
OpenWrt 19.07.6 still seems to package a statically linked getdns_query.

izstas, 2021-04-24:
I can confirm that this has been fixed in both 19.07.7 and 21.02-rc1. Thanks!

closed 2021-04-24 by izstas