openwrt/packages #13383: samba4: compile stopped at

youxiaojie, 2020-09-13:
```
make[3]: Entering directory '/home/allan/openwrt/feeds/packages/net/samba4'
(cd /home/allan/openwrt/build_dir/hostpkg/samba-4.12.5/; if [ -x configure ]; then cp -fpR /home/allan/openwrt/scripts/config.{guess,sub} /home/allan/openwrt/build_dir/hostpkg/samba-4.12.5// && CC="gcc" CFLAGS="-O2 -I/home/allan/openwrt/staging_dir/host/include -I/home/allan/openwrt/staging_dir/hostpkg/include -I/home/allan/openwrt/staging_dir/target-arm_mpcore_musl_eabi/host/include" CXX="g++" CPPFLAGS="-I/home/allan/openwrt/staging_dir/host/include -I/home/allan/openwrt/staging_dir/hostpkg/include -I/home/allan/openwrt/staging_dir/target-arm_mpcore_musl_eabi/host/include" LDFLAGS="-L/home/allan/openwrt/staging_dir/host/lib -L/home/allan/openwrt/staging_dir/hostpkg/lib -L/home/allan/openwrt/staging_dir/target-arm_mpcore_musl_eabi/host/lib" CONFIG_SHELL="/usr/bin/env bash" PATH="/home/allan/openwrt/staging_dir/hostpkg/bin:/home/allan/openwrt/staging_dir/hostpkg/usr/bin:/home/allan/openwrt/staging_dir/toolchain-arm_mpcore_gcc-8.4.0_musl_eabi/bin:/home/allan/openwrt/staging_dir/toolchain-arm_mpcore_gcc-8.4.0_musl_eabi/bin:/home/allan/openwrt/staging_dir/host/bin:/home/allan/openwrt/staging_dir/toolchain-arm_mpcore_gcc-8.4.0_musl_eabi/bin:/home/allan/openwrt/staging_dir/host/bin:/home/allan/openwrt/staging_dir/host/bin:/home/allan/bin:/home/allan/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"  ./buildtools/bin/waf configure --prefix=/home/allan/openwrt/staging_dir/hostpkg --exec-prefix=/home/allan/openwrt/staging_dir/hostpkg --sysconfdir=/home/allan/openwrt/staging_dir/hostpkg/etc --localstatedir=/home/allan/openwrt/staging_dir/hostpkg/var --sbindir=/home/allan/openwrt/staging_dir/hostpkg/bin --hostcc="gcc" --disable-cups --disable-iprint --disable-cephfs --disable-fault-handling --disable-glusterfs --disable-spotlight --disable-rpath --disable-rpath-install --disable-rpath-private-install --enable-fhs --without-automount --without-iconv --without-lttng --without-ntvfs-fileserver --without-pam --without-systemd --without-utmp --without-dmapi --without-fam --without-gettext --without-regedit --without-gpgme --disable-avahi --without-quotas --without-acl-support --without-winbind --without-ad-dc --without-json --without-libarchive --disable-python --nopyc --nopyo --without-dnsupdate --without-ads --without-ldap --without-ldb-lmdb --with-static-modules=!DEFAULT,!FORCED --with-shared-modules=!DEFAULT,!FORCED --builtin-libraries=replace --nonshared-binary=asn1_compile,compile_et ; fi )
Setting top to                           : /home/allan/openwrt/build_dir/hostpkg/samba-4.12.5 
Setting out to                           : /home/allan/openwrt/build_dir/hostpkg/samba-4.12.5/bin 
Checking for 'gcc' (C compiler)          : gcc 
Checking for program 'git'               : /home/allan/openwrt/staging_dir/host/bin/git 
Checking for c flags '-MMD'              : yes 
Checking for program 'gdb'               : /usr/bin/gdb 
Checking for header sys/utsname.h        : yes 
Checking uname sysname type              : Linux 
Checking uname machine type              : i686 
Checking uname release type              : 4.15.0-117-generic 
Checking uname version type              : #118~16.04.1-Ubuntu SMP Sat Sep 5 23:29:42 UTC 2020 
Checking for header stdio.h              : yes 
Checking simple C program                : ok 
Checking compiler accepts ['-Werror']    : yes 
Checking linker accepts ['-Wl,-rpath,.'] : yes 
Checking for rpath library support       : yes 
Checking for -Wl,--version-script support : yes 
Checking compiler accepts ['-fvisibility=hidden'] : yes 
Checking for HAVE_VISIBILITY_ATTR                 : ok 
Checking for library constructor support          : ok 
Checking for library destructor support           : ok 
Checking for __attribute__                        : ok 
Checking compiler accepts ['-fPIC']               : yes 
Checking for inline                               : inline 
Checking for program 'pkg-config'                 : /home/allan/openwrt/staging_dir/host/bin/pkg-config 
Checking for pkg-config version >= '0.0.0'        : yes 
Checking compiler accepts ['-D__STDC_WANT_LIB_EXT1__=1'] : yes 
Checking compiler accepts ['']                           : yes 
Checking for header sys/types.h                          : yes 
Checking for header sys/stat.h                           : yes 
Checking for header stdlib.h                             : yes 
Checking for header stddef.h                             : yes 
Checking for header memory.h                             : yes 
Checking for header string.h                             : yes 
Checking for header strings.h                            : yes 
Checking for header inttypes.h                           : yes 
Checking for header stdint.h                             : yes 
Checking for header unistd.h                             : yes 
Checking for header minix/config.h                       : no 
Checking for header ctype.h                              : yes 
Checking for header standards.h                          : no 
Checking for header stdbool.h                            : yes 
Checking for header stdarg.h                             : yes 
Checking for header vararg.h                             : no 
Checking for header limits.h                             : yes 
Checking for header assert.h                             : yes 
Checking getconf LFS_CFLAGS                              : -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 
Checking getconf large file support flags work           : ok 
Checking for large file support without additional flags : ok 
Checking for header sys/time.h                           : yes 
Checking for header time.h                               : yes 
Checking for header endian.h                             : yes 
Checking for header sys/endian.h                         : no 
Checking for HAVE_LITTLE_ENDIAN                          : ok 
Checking for HAVE_BIG_ENDIAN                             : not found 
Checking for header signal.h                             : yes 
Checking if signal handlers return int                   : not found 
Checking for variable __FUNCTION__                       : ok 
Checking for va_copy                                     : ok 
Checking for HAVE__VA_ARGS__MACRO                        : ok 
Checking for header linux/types.h                        : yes 
Checking for header crypt.h                              : yes 
Checking for header locale.h                             : yes 
Checking for header acl/libacl.h                         : no 
Checking for header compat.h                             : no 
Checking for header attr/xattr.h                         : no 
Checking for header dustat.h                             : no 
[... 939 lines cut ...]
Using flex version                                                                     : flex 2.6.4 
Checking for header rpc/xdr.h                                                          : yes 
Checking for library nscd                                                              : no 
Checking for nscd_flush_cache                                                          : not found 
VFS_STATIC: vfs_default,vfs_not_implemented
VFS_SHARED: 
PDB_STATIC: 
PDB_SHARED: 
AUTH_STATIC: 
AUTH_SHARED: 
NSS_INFO_STATIC: 
NSS_INFO_SHARED: 
CHARSET_STATIC: 
CHARSET_SHARED: 
IDMAP_STATIC: 
IDMAP_SHARED: 
GPEXT_STATIC: 
GPEXT_SHARED: 
PERFCOUNT_STATIC: 
PERFCOUNT_SHARED: 
RPC_STATIC: 
RPC_SHARED: 
Checking for openpty                                                                   : 

```

Andy2244, 2020-09-14:
Not sure, never seen it stop there.
Try clean SDK/Build and if this fails might be a Distro issue with samba. I use Debian + WSL2 for my builds and my docker-builder also uses Debian (debian:stretch-slim).

Neustradamus, 2021-04-13:
@youxiaojie: It has been solved?

Neustradamus, 2021-09-12:
@youxiaojie: Any news?