openwrt/packages #13395: CI: super-linter does not understand OpenWRT shell scripting paradigms

aaronjg, 2020-09-15:
@aparcar @neheb 

Issue with the recent change made here: #13358

I have been working on  P/R #13169, and saw that there is now a new delinter in the CI process. While, delinters are often useful for maintaining code quality. the shellcheck plugin to super linter has some major issues for OpenWRT code:

1) It is unable to find sourced scripts (SC1091), which causes unused variable warnigs (SC2039)
2) It doesn't seem to understand how variables are assigned using config_get, causing many unused variable warnings (SC2039)
3) It does not understand the capabilities of busybox ash causing many SC2039 warnings
4) It gives lots of false positives for SC2086 (double quoting). Often variables are set using functions/network.sh or other scripts that will not have spaces, so do not need to be double quoted.

So many false positives will just train committers to ignore all the shellcheck warnings. Delinter should be turned off for shell scripts until it can be made to better understand the OpenWRT environment and shell scripting paradigm.

aparcar, 2020-09-15:
Thank you very much for the quick feedback, I suggest to add the following env to the workflow file:

```shell
SHELLCHECK_OPTS: "-e SC1091 -e SC2039"
```

> It gives lots of false positives for SC2086 (double quoting). Often variables are set using functions/network.sh or other scripts that will not have spaces, so do not need to be double quoted.

Better safe than sorry I'd say, don't you think?

aparcar, 2020-09-15:
Maybe like this? https://github.com/openwrt/packages/pull/13396

neheb, 2020-09-15:
I wouldn't worry too much. Shellcheck gives valuable feedback. All the linter does from my point of view is to prevent merging on mobile.

The quoting warnings are annoying since they can result in bugs, especially when used with procd functions.

aaronjg, 2020-09-15:
We need to be able to source the scripts, rather than just ignoring the warning that the scripts aren't sourced.  Turning off SC2034 is going to cause a lot of false negatives, when variables are actually unused. (sorry, had SC2039 for that in my initial comment, and have edited it now).

SC2039 could probably be turned off, but it would be nice if it could still alert if a user used a function that was not available in `ash`.

If this is going to be useful, it needs to have a somewhat reasonable false positive rate. As far as I can tell, I've taken care of all of the relevant shellcheck warnings for mwan3, but it is still giving about 2,500 lines of messages.

neheb, 2020-09-18:
If we look at https://github.com/openwrt/packages/blob/master/net/nut/files/nut_serial.hotplug#L23 , we see there's a variable for this.

I don't know if this will work for the environment that GitHub Actions uses.

neheb, 2020-09-18:
Hmmm there's no easy solution to this. I was thinking having shellcheck being ran, similar to test.sh but shellcheck is written in haskell.

aaronjg, 2020-09-18:
> If we look at https://github.com/openwrt/packages/blob/master/net/nut/files/nut_serial.hotplug#L23 , we see there's a variable for this.
I don't know if this will work for the environment that GitHub Actions uses.

Yeah, it would be ideal if there was an environment variable or some other way to tell it where to find the other scripts to include.

jefferyto, 2020-10-06:
I think we can add stub include files like `/lib/functions.sh` into the linting environment (with empty variable/function definitions, because further linting of OpenWrt core shell scripts may lead to more linting errors), and preprocess the target scripts, e.g. replace calls to `config_get` with variable definitions.

But I agree with @aaronjg - blindly ignoring/disabling shellcheck errors is not the answer and this linting should be turned off until further work is done to make it work with OpenWrt code. (And to be clear there is a lot of work that needs to be done here.) Leaving this linting check enabled will deter/discourage smaller changes from casual contributors.

At the very least shellcheck should have been run on all existing shell code in this repo to see the potential impact before #13358 was merged. ["Just merge and see how people like it?"](https://github.com/openwrt/packages/pull/13358#issuecomment-692309158) is frankly just irresponsible.

neheb, 2020-10-06:
Some other projects like aports have a linter that is marked allow to fail. So there's that.

The core OpenWrt sh files also need linting badly. Upstream doesn't seem to care much though.

jefferyto, 2020-10-06:
> Leaving this linting check enabled will deter/discourage smaller changes from casual contributors.

I wanted to clarify this: My guess is that most (if not all) init and other on-target scripts here do not currently pass shellcheck. If a casual contributor wanted to fix a small bug in a script, their PR would not pass the linter and they would be on the hook to fix all of the shellcheck errors in that script. I would imagine many would abandon their PRs at this point.

> Some other projects like aports have a linter that is marked allow to fail. So there's that.

I am ambivalent about this option. On the one hand, this allows valid PRs to be merged without having to fix linting errors. On the other hand, this means there is no incentive for people to investigate the linting results and improve their code.

neheb, 2020-10-06:
> I wanted to clarify this: My guess is that most (if not all) init and other on-target scripts here do not currently pass shellcheck. If a casual contributor wanted to fix a small bug in a script, [...]

Actually  that only applies to mobile devices. On desktop, I can merge failing PRs.

> On the other hand, this means there is no incentive for people to investigate the linting results and improve their code.

I assume that’s what the review process is for.

BKPepe, 2020-10-17:
> At the very least shellcheck should have been run on all existing shell code in this repo to see the potential impact before #13358 was merged. ["Just merge and see how people like it?"](https://github.com/openwrt/packages/pull/13358#issuecomment-692309158) is frankly just irresponsible.

🔝 

> On desktop, I can merge failing PRs.

Why would you do that in the first place?

neheb, 2020-10-17:
There are known problematic failures.

Boost does not compile under CircleCI.

libtalloc does not compile under GitHub Actions but does under the buildbots since they have distutils.

Linting is mostly cosmetic.

EricLuehrsen, 2020-11-04:
> At the very least shellcheck should have been run on all existing shell code in this repo to see the potential impact before #13358 was merged. "Just merge and see how people like it?" is frankly just irresponsible.

Lets consider many of our day jobs. When harsh quality rules are agreed by a panel of experts and management they are still contentious through out an organization. A shoot from the hip commit like #13358 would get a person BBQ'd even on a small team. Why would this be inflicted on an open source "extra" packages repository with diverse contributors?

Shellcheck should be disabled. It is pedantic and has no structure in its alleged error codes. It should have a setting for POSIX plus common sense. For example, `local` keyword is strongly encouraged in most places but not POSIX. Most lint tools have error codes in some kind of order, such as numbered lowest is error and highest is just info. Therefore a switch can be set at a cut off number. As mentioned, shellcheck is run preinstall so none of the sourcing works, and a lone stub functions.sh won't fix that. (fyi this paragraph is more than 400 characters, see next paragraph).

Markdownlinter is useful before you publish and can help you find your mistakes, if you don't find them yourself. The ultimate test however is to load your MD into the viewer environment and make a **qualitative** judgement. Has your information been presented how you want it to be? I have had the same instruction guide for years, so what is this 400 character limit nonsense now? If you want a paragraph viewer-justified without odd line breaks as a browser shifts, then it must be a run-on one-liner.

I am now trying to do a minor version update and a few bug fixes on Unbound and this stupid thing is giving me unnecessary grief.

EricLuehrsen, 2020-11-04:
> At the very least shellcheck should have been run on all existing shell code in this repo to see the potential impact before #13358 was merged. "Just merge and see how people like it?" is frankly just irresponsible.

Lets consider many of our day jobs. When harsh quality rules are agreed by a panel of experts and management they are still contentious through out an organization. A shoot from the hip commit like #13358 would get a person BBQ'd even on a small team. Why would this be inflicted on an open source "extra" packages repository with diverse contributors?

Shellcheck should be disabled. It is pedantic and has no structure in its alleged error codes. It should have a setting for POSIX plus common sense. For example, `local` keyword is strongly encouraged in most places but not POSIX. Most lint tools have error codes in some kind of order, such as numbered lowest is error and highest is just info. Therefore a switch can be set at a cut off number. As mentioned, shellcheck is run preinstall so none of the sourcing works, and a lone stub functions.sh won't fix that.

Markdownlinter is useful before you publish and can help you find your mistakes, if you don't find them yourself. The ultimate test however is to load your MD into the viewer environment and make a qualitative judgement. Has your information been presented how you want it to be? I have had the same instruction guide for years, so what is this 400 character limit now? If you want a paragraph viewer-justified without odd line breaks as a browser shifts, then it must be a run-on one-liner.  (BTW this paragraph is more than 400 characters).

I am now trying to do a minor version update and a few bug fixes on Unbound and this is giving me unnecessary grief.

aaronjg, 2020-11-06:
Sounds like many users do not like this. Since the initial motivation was "Just merge and see how people like it?", that should be enough reason to revert this until we can have a better solution.

P/R to remove the linter: https://github.com/openwrt/packages/pull/13861

closed 2020-11-06 by aaronjg