openwrt/packages #13408: Udev

predators46, 2020-09-17:
whether eudev is udev whose name was changed, and whether the function is the same as udev

neheb, 2020-09-19:
currently, there is libudev-fbsd in here, which is mostly a stub library based on older udev. There's work on introducing a better one: https://github.com/openwrt/packages/issues/12844

In any case, udev or eudev will not be coming here.

closed 2020-09-19 by neheb

predators46, 2020-09-20:
> saat ini, ada libudev-fbsd di sini, yang sebagian besar merupakan pustaka rintisan berdasarkan udev lama. Ada upaya untuk memperkenalkan yang lebih baik: [# 12844](https://github.com/openwrt/packages/issues/12844)
> 
> Bagaimanapun, udev atau eudev tidak akan datang ke sini.

but I managed to make it in the master branch

predators46, 2020-09-20:
> currently, there is libudev-fbsd in here, which is mostly a stub library based on older udev. There's work on introducing a better one: #12844
> 
> In any case, udev or eudev will not be coming here.

but I managed to make it in the master branch