openwrt/packages #13424: mwan3 - could there be a stick on when its working again on snapshots?

thomaslunze, 2020-09-18:
I know that mwan3 is not yet ready to be working with 20.x releases right now (as per wiki) but asking if ther is a stick note already somewhere or could be created to just knwo when it amakes sense to build another snapshot with imagebuilder and have a working mwan3 again?

my current "OpenWrt SNAPSHOT r14154" works fine.
anything beyond does not

Thanks everyone

jamesmacwhite, 2020-09-18:
Best thing to do is keep an eye on some of the open issues here currently: https://github.com/openwrt/packages/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+mwan3

This WIP PR from @aaronjg is one of the first major steps in working towards mwan3 working on snapshot: https://github.com/openwrt/packages/pull/13169 and resolves some of the key issues, but more testing and development is likely needed in other areas too.

The documentation will be updated as the codebase changes, so you are best to keep checking their and here on GitHub really to know when developments occur.

aaronjg, 2020-09-18:
You can build the P/R now, but it will require either imagebuilder + SDK or buildroot. P/R is essentially done. I, and other users who have tested it on the snapshot build are reporting that it works.

Out of curiosity, what is the date of the snapshot build, router architecture,  and output of `uname -r` for the version that it works on?

thomaslunze, 2020-09-19:
The working snapshot I am running I have buildroot on 13.08.2020

details:
Hostname	OpenWrt
Model	Linksys WRT1200AC
Architecture	ARMv7 Processor rev 1 (v7l)
Firmware Version	OpenWrt SNAPSHOT r14154-2d9db93397 / LuCI Master git-20.219.57394-35e2735

root@OpenWrt:~# uname -r
5.4.52

jamesmacwhite, 2020-09-22:
Once 2.9.10 is merged into master it should improve the situation for snapshot but as @aaronjg said you can build the PR now with the SDK if you want. I added the big red banner on the docs because a few reports were coming in here with the same issue. Because of the requirement to build with the SDK currently until it's released, I wouldn't expect everyone to compile the PR branch, so that's why it's warned against currently.

thomaslunze, 2020-09-22:
Thanks everyone. As I am not that familiar with building the PR I will wait for the merge to the master :-)
Or is there any easy hint on how to do that? *fearless*

jamesmacwhite, 2020-09-22:
If you fancy it, you can look at the SDK guide here: https://openwrt.org/docs/guide-developer/using_the_sdk. You don't need to go with a full compile, the SDK removes a lot of that as it's pre-compiled toolchain, rather than building from source entirely, but you will still need to compile the mwan3 PR branch yourself with the SDK.

The PR should be ready soon though, if you can wait. I would stay on the snapshot build that works, once the PR gets merged you'll then be able to upgrade using opkg from the snapshot repo for your router within 24 hours or so, once the package gets built, then you'll be able to upgrade to another snapshot build and things should still work under 2.9.10.

thomaslunze, 2020-10-09:
its still not merged ?

aaronjg, 2020-10-09:
Florian identified some issues with disconnected events not being triggered, and I am still working to sort this out. I believe once those are fixed, it can be merged.

thomaslunze, 2020-10-09:
awesome and thanks for the update :-) and all your hard work on this project

aaronjg, 2020-10-21:
The P/R has been merged, and should be available in tomorrow's snapshot build.

thomaslunze, 2020-10-21:
awesome guys. thanks for your hard work

closed 2020-10-21 by thomaslunze