openwrt/packages #13443: mwan3: /sbin/hotplug-call: eval: line 847: continuea: not found

elcovv, 2020-09-20:
Maintainer: @\Florian Eckert
Environment: ar71xx/generic, tl-wr902ac-v1, 19.07.4

Description:
When I restart the mwan3 service or whenever mwan3 acts on a hotplug event I get the following error messages on the active console:

root@TL-WR902ACv1:~# service mwan3 restart
/sbin/hotplug-call: eval: line 847: continuea: not found
/sbin/hotplug-call: eval: line 847: continuea: not found
root@TL-WR902ACv1:~#

looks like a typo in line 847 of /lib/mwan3/mwan3.sh.

Regards,

Elco

jamesmacwhite, 2020-09-20:
Looks like a typo did slip in:

https://github.com/openwrt/packages/blob/ab747fe0fb3b60c0bd42ea2c4fc045273eb46f6f/net/mwan3/files/lib/mwan3/mwan3.sh#L847

This will have been fixed in master, due to refactoring done in 2.9.0 and this is no longer present.

You can upgrade from the snapshot package for your arch: https://downloads.openwrt.org/snapshots/packages/ or temporarily edit the `/lib/mwan3/mwan3.sh` until it is made in the openwrt-19.07 branch.

jamesmacwhite, 2020-09-21:
Just pinging @feckert as the original issue didn't tag properly.

feckert, 2020-09-21:
@jamesmacwhite Thanks for looking into this.

closed 2020-09-21 by feckert

aaronjg, 2020-09-22:
Thanks for catching that. This just requires a small change that can be applied to 19.07. Since this is on the stable branch, I'd like to get this fixed ASAP, and it may be a couple days before I have time to address @feckert's comments on the other PR.

jamesmacwhite, 2020-09-24:
@elcovv FYI mwan3 2.8.13 is now live in 19.07 with this fixed, so you can upgrade via opkg/luci

PR #13466 (2020-09-23): mwan3: fix typo in mwan3_set_sticky_iptables [19.07] [bugfix]
fixes #13443
commit 49459505 (2020-09-23): mwan3: fix typo in mwan3_set_sticky_iptables

fixes #13443