openwrt/packages #13469: ttyd: closes session on firefox after 30 seconds

feckert, 2020-09-22:
Maintainer: @tsl0922 
Environment: x86_64, APU3, OpenWrt master

Description:

I have installed luci-app-ttyd from master.
I have set the following configuration.

```
config ttyd 'local'
        option interface '@lan'
        option command '/usr/libexec/login.sh'
        option ssl '1'
        option ssl_cert '/etc/uhttpd.pem'
        option ssl_key '/etc/uhttpd.key'
```
No matter if I am logged in or not, after 30 seconds the session is closed and restarted on firefox.
I do not have this behaviour with chromium.

Version:
Chromium: Version 83.0.4103.116 (Developer Build) built on Debian 10.4, running on Deb
Firefox: 68.10.0esr.

Logging:
```
Tue Sep 22 15:40:19 2020 daemon.notice ttyd[12553]: HTTP /token - 192.168.0.2
Tue Sep 22 15:40:19 2020 daemon.notice ttyd[12553]: WS   /ws - 192.168.0.2, clients: 1
Tue Sep 22 15:40:19 2020 daemon.notice ttyd[12553]: started process, pid: 13414
Tue Sep 22 15:40:46 2020 daemon.notice ttyd[12553]: WS closed from 192.168.0.2, clients: 0
Tue Sep 22 15:40:46 2020 daemon.notice ttyd[12553]: killing process 13414 with signal: 1 (SIGHUP)
Tue Sep 22 15:40:46 2020 daemon.notice ttyd[12553]: process killed with signal 1 (SIGHUP), pid: 13414
Tue Sep 22 15:41:19 2020 daemon.notice ttyd[12553]: HTTP /token - 192.168.0.2
Tue Sep 22 15:41:19 2020 daemon.notice ttyd[12553]: WS   /ws - 192.168.0.2, clients: 1
Tue Sep 22 15:41:19 2020 daemon.notice ttyd[12553]: started process, pid: 13451
Tue Sep 22 15:41:46 2020 daemon.notice ttyd[12553]: WS closed from 192.168.0.2, clients: 0
Tue Sep 22 15:41:46 2020 daemon.notice ttyd[12553]: killing process 13451 with signal: 1 (SIGHUP)
Tue Sep 22 15:41:46 2020 daemon.notice ttyd[12553]: process killed with signal 1 (SIGHUP), pid: 13451
```

Eszartek, 2020-10-22:
I have the exact same issue with Firefox only. I'm running Firefox 81.0.2 64-bit for Arch Linux. I have the Midori 9.0 browser and ttyd does not time out after 30 seconds.

Eszartek, 2020-10-22:
update: On the assumption this is a Firefox bug, I figured I'd check out the websocket settings in about:config.

I changed network.websocket.timeout.close from 20 to 200 and it seemed to allow ttyd to run longer. I then changed the value to 5 and ttyd continued to work for over 10 minutes, and its still going. So there may be a Firefox timeout bug that is mitigated by changing a default value to some other value. I'll continue to test.

allista, 2023-10-09:
The same happens to me, but on google-chrome

The `ttyd` is running behind `nginx` reverse proxy, but no amount of `proxy_*_timeout` options change it in any way.