openwrt/packages #13474: strongswan: wolfssl plugin package

yuan0qi, 2020-09-23:
Backend has been added since version 5.8.0 and provides RSA/ECDSA/DH/ECDH/X25519/Ed25519/ciphers/hashers/HMAC/RNG

derekyerger, 2020-11-25:
On target x86_64 I'm seeing:

`
OpenWrt-libtool: compile:  x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../../.. -I../../../../src/libstrongswan -I/home/derek/bak/openwrt/staging_di
r/target-x86_64_musl/usr/include -I/home/derek/bak/openwrt/staging_dir/target-x86_64_musl/include -I/home/derek/bak/openwrt/staging_dir/toolchain-x86_64_gcc-7.
5.0_musl/usr/include -I/home/derek/bak/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/derek/bak/openwrt/staging_dir/toolchain-x86_
64_gcc-7.5.0_musl/include -I/home/derek/bak/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/derek/bak/openwrt/staging_dir/target-x
86_64_musl/usr/lib/libintl-stub/include -rdynamic -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-resu
lt -iremap/home/derek/bak/openwrt/build_dir/target-x86_64_musl/strongswan-5.8.2:strongswan-5.8.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_
SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/derek/bak/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/derek/bak/openwrt/staging_dir/t
arget-x86_64_musl/usr/lib/libintl-stub/include -include /home/derek/bak/openwrt/build_dir/target-x86_64_musl/strongswan-5.8.2/config.h -MT wolfssl_rsa_public_k
ey.lo -MD -MP -MF .deps/wolfssl_rsa_public_key.Tpo -c wolfssl_rsa_public_key.c  -fPIC -DPIC -o .libs/wolfssl_rsa_public_key.o
In file included from ../../../../src/libstrongswan/credentials/certificates/certificate.h:30:0,
                 from ../../../../src/libstrongswan/credentials/credential_set.h:28,
                 from ../../../../src/libstrongswan/credentials/credential_manager.h:31,
                 from ../../../../src/libstrongswan/library.h:113,
                 from wolfssl_common.h:26,
                 from wolfssl_rsa_public_key.c:23:
../../../../src/libstrongswan/credentials/keys/signature_params.h:109:34: error: expected identifier before '-' token
 #define RSA_PSS_SALT_LEN_DEFAULT -1
                                  ^
make[9]: *** [Makefile:617: wolfssl_rsa_public_key.lo] Error 1
make[9]: Leaving directory '/home/derek/bak/openwrt/build_dir/target-x86_64_musl/strongswan-5.8.2/src/libstrongswan/plugins/wolfssl'
`

The same result is seen when using glibc.

Patch from fb9077a:
`
diff --git a/net/strongswan/Makefile b/net/strongswan/Makefile
index 89c0d72..aa1c2fe 100644
--- a/net/strongswan/Makefile
+++ b/net/strongswan/Makefile
@@ -89,6 +89,7 @@ PKG_MOD_AVAILABLE:= \
        updown \
        vici \
        whitelist \
+       wolfssl \
        x509 \
        xauth-eap \
        xauth-generic \
@@ -211,6 +212,7 @@ $(call Package/strongswan/Default)
        +strongswan-mod-updown \
        +strongswan-mod-vici \
        +strongswan-mod-whitelist \
+       +strongswan-mod-wolfssl \
        +strongswan-mod-x509 \
        +strongswan-mod-xauth-eap \
        +strongswan-mod-xauth-generic \
@@ -586,6 +588,12 @@ define Plugin/whitelist/install
        $(CP) $(PKG_INSTALL_DIR)/usr/lib/ipsec/plugins/libstrongswan-whitelist.so $(1)/usr/lib/ipsec/plugins/
 endef
 
+define Plugin/wolfssl/install
+       $(INSTALL_DIR) $(1)/usr/lib/ipsec/plugins
+       $(CP) $(PKG_INSTALL_DIR)/usr/lib/ipsec/wolfssl $(1)/usr/lib/ipsec/
+       $(CP) $(PKG_INSTALL_DIR)/usr/lib/ipsec/plugins/libstrongswan-wolfssl.so $(1)/usr/lib/ipsec/plugins/
+endef
+
 define Plugin/kernel-libipsec/install
        $(INSTALL_DIR) $(1)/usr/lib/ipsec
        $(CP) $(PKG_INSTALL_DIR)/usr/lib/ipsec/libipsec.so.* $(1)/usr/lib/ipsec/
@@ -673,6 +681,7 @@ $(eval $(call BuildPlugin,unity,Cisco Unity extension,))
 $(eval $(call BuildPlugin,updown,updown firewall,))
 $(eval $(call BuildPlugin,vici,Versatile IKE Configuration Interface,))
 $(eval $(call BuildPlugin,whitelist,peer identity whitelisting,))
+$(eval $(call BuildPlugin,wolfssl,WolfSSL crypto,+PACKAGE_strongswan-mod-wolfssl:libwolfssl))
 $(eval $(call BuildPlugin,x509,x509 certificate,))
 $(eval $(call BuildPlugin,xauth-eap,EAP XAuth backend,))
 $(eval $(call BuildPlugin,xauth-generic,generic XAuth backend,))
`

derekyerger, 2020-11-27:
Added PR #14043