openwrt/packages #13484: rtl_433 not deconding properly on some platforms

ouinouin, 2020-09-24:
hi, 
i own many openwrt platform and begin to run rtl433 on them.

on 
Model
TP-LINK Archer C7 v5
Architecture
Qualcomm Atheros QCA956X ver 1 rev 0
Firmware Version
OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.136.49537-fb2f363
Kernel Version
4.14.180
I have similar decoding than on my laptop (i own many of those probes : 
```
time      : 2020-09-24 12:38:59
model     : Ambientweather-F007TH                  House Code: 103
Channel   : 3            Battery   : 1             Temperature: 75.4 F       Humidity  : 59 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-24 12:39:12
model     : Ambientweather-F007TH                  House Code: 89
Channel   : 2            Battery   : 1             Temperature: 74.6 F       Humidity  : 60 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-24 12:39:13
model     : Ambientweather-F007TH                  House Code: 11
Channel   : 1            Battery   : 1             Temperature: 76.0 F       Humidity  : 62 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-24 12:39:17
model     : Ambientweather-F007TH                  House Code: 124
Channel   : 7            Battery   : 1             Temperature: 75.3 F       Humidity  : 60 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-24 12:39:19
model     : Ambientweather-F007TH                  House Code: 171
Channel   : 2            Battery   : 1             Temperature: 76.6 F       Humidity  : 63 %          Integrity : CRC

```
on 
Model
Netgear Nighthawk X4S R7800
Architecture
ARMv7 Processor rev 0 (v7l)
Firmware Version
OpenWrt 19.07.4 r11208-ce6496d796 / LuCI openwrt-19.07 branch git-20.247.75781-0d0ab01
Kernel Version
4.14.195

its a disaster, most of the time it listen nothing (all 433 sensors are within 5 meters with battery ok and no obstacles and the same test with my laptop is ok) when it decides to see something, the decoding gets wrong :
```

time      : 2020-09-24 12:48:32
Model     : Oregon-SL109H Id        : 0
Channel   : 3            Celsius   : 0.00 C        Humidity  : 0 %           Status    : 0             Integrity : CHECKSUM
```
i got this bogus result, and  while seeing the transmission led of my 433mhz temp sensor I can ensure this is a frame given by my sensor, wich is Ambientweather-F007TH.

i did further tests on the faulty router : i got rtl tcp running and the decoding done on another platform (X86-64 debian) and got the same bogus results so there is something wrong in the rtl driver, in the timings of the router, or in the cpufreq giving some timer issues or in the I/Q endianness or something like that.

neheb, 2020-12-22:
I may or may have figured this out while cleaning up the Makefile.

rtl_433 comes with a bunch of configuration files that are not being installed. Do you think that's the issue?

ouinouin, 2020-12-22:
i dont think so, to what i know the config files are just for a few kind of sensors, thos i own are very chineric ( as chinese generic). i m more thinking about something about timers , dual core issue interrupts or things like that but thats just guessings. on archer c7 openwr the same  rtl433 package is working properly with the same probes.

neheb, 2020-12-22:
Oh. I just saw R7800...

That thing when I owned it had great WiFi performance, but everything else was subtly broken. I would suggest setting cpufreq (or w/e it was)to performance and see if that happens.

Maybe @Ansuel and @hnyman can help.

On a somewhat unrelated sidenote, Armbian carried a totally broken out of tree cpufreq patch for mvebu that caused kernel freezes every hour or so. Thank god I got it removed.

ouinouin, 2020-12-22:
i think i already gave a try for that but cant remember, i ll have a look as soon as i m near the router (its on a remote site)

robho, 2023-05-06:
I think this has been fixed in rtl_433: https://github.com/merbanan/rtl_433/issues/1441

GrumpyMeow, 2025-01-13:
Can this issue be closed? If so.. please close.

closed 2025-01-13 by neheb