openwrt/packages #13492: mwan3: Not re-routing to backup interface

brianjmurrell, 2020-09-25:
Maintainer: @feckert 
Environment: x86, generic, 19.07.3

Description:
I have a primary/backup configuration but when the primary connection (`eth0.2`) loses pings, routing is not adjusted to use the backup connection (`pppoe-wan1`).

The troubleshooting tab shows:
```
Software-Version
-------------------------------------------------
OpenWrt - OpenWrt 19.07.3 r11063-85e04e9f46
LuCI - git-20.221.53167-304ad04

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
3: ifb0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 12:16:bd:cf:63:e5 brd ff:ff:ff:ff:ff:ff
4: ifb1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 0e:32:71:bf:3d:38 brd ff:ff:ff:ff:ff:ff
5: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:08:74:96:6d:9b brd ff:ff:ff:ff:ff:ff
    inet6 fe80::208:74ff:fe96:6d9b/64 scope link 
       valid_lft forever preferred_lft forever
6: br-guest: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 00:08:74:96:6d:9b brd ff:ff:ff:ff:ff:ff
    inet 192.168.101.251/24 brd 192.168.101.255 scope global br-guest
       valid_lft forever preferred_lft forever
    inet 192.168.101.254/24 brd 192.168.101.255 scope global secondary br-guest
       valid_lft forever preferred_lft forever
    inet6 fe80::208:74ff:fe96:6d9b/64 scope link 
       valid_lft forever preferred_lft forever
7: eth0.15@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-guest state UP group default qlen 1000
    link/ether 00:08:74:96:6d:9b brd ff:ff:ff:ff:ff:ff
8: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 00:08:74:96:6d:9b brd ff:ff:ff:ff:ff:ff
    inet 10.75.22.251/24 brd 10.75.22.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet 10.75.22.254/24 brd 10.75.22.255 scope global secondary br-lan
       valid_lft forever preferred_lft forever
    inet6 2607:abcd:9365:8300::1/60 scope global dynamic noprefixroute 
       valid_lft 62938sec preferred_lft 62938sec
    inet6 2001:123:ab:123::1/64 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fd31:aeb1:48df::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::208:74ff:fe96:6d9b/64 scope link 
       valid_lft forever preferred_lft forever
9: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 00:08:74:96:6d:9b brd ff:ff:ff:ff:ff:ff
10: eth0.3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 00:08:74:96:6d:9b brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.251/24 brd 10.0.0.255 scope global eth0.3
       valid_lft forever preferred_lft forever
    inet6 fe80::208:74ff:fe96:6d9b/64 scope link 
       valid_lft forever preferred_lft forever
35: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none 
    inet 10.8.0.1/24 brd 10.8.0.255 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fd8c:5604:c29c:c562::1/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::1f7c:133:d00c:1eeb/64 scope link stable-privacy 
[... 344 lines cut ...]
config policy 'wan1_henet_wan0'
	list use_member 'wan0_m2_w3'
	list use_member 'wan1_m1_w2'
	#list use_member 'wan0_6_m2_w3'
	list use_member 'henet_m2_w3'
	list use_member 'wan1_6_m1_w2'

config rule 'TS_NNTP'
	option dest_port '119'
	option proto 'tcp'
	option sticky '0'
	option use_policy 'wan1_only'
	option dest_ip '1.2.3.4/31'
	option family 'ipv4'

config rule 'TS_NNTP_v6'
	option dest_ip '2001:1:2:3::20/127'
	option dest_port '119'
	option proto 'tcp'
	option sticky '0'
	option use_policy 'wan1_only'
	option family 'ipv6'
```

If the cause is not obvious, I will start debugging, but I was hoping it would be obvious and could save me some time.

brianjmurrell, 2020-09-25:
I forget much of how this stuff is configured, but shouldn't:

```
config rule 'default_rule'
	option dest_ip '0.0.0.0/0'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan0_wan1'
```

mean there should be a:

```
config policy 'wan0_wan1'
```
?

brianjmurrell, 2020-09-25:
I think what I probably want[ed] was:

```
config rule 'default_rule'
	option dest_ip '0.0.0.0/0'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan0_henet_wan1'
```

I must say however, I find it difficult to wrap my mind around there being just a single rule for the default route for both ipv4 and ipv6.  Historically, when I have done this kind of multi-wan configuration (using tools other than mwan3) I have treated ipv4 and ipv6 completely separate with a parallel set of configs, one for each.

In any case, changing as per the above still does not result in a default route via `pppoe_wan1` when `eth0.2` goes down.

jamesmacwhite, 2020-09-25:
@brianjmurrell IPv4 and IPv6 being separate was something that was changed in the default config a few versions ago. As I agree it is quite weird having a IPv4 mask rule controlling both IPv4 and IPv6. It is now like this:

https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/config/mwan3#L141-L149

Other experts would need to look at the failover configuration in more detail, but just wanted to point out the configuration change.

brianjmurrell, 2020-09-25:
So what's the theory of operation for failover routing?  Let's take the simple case of "balanced", shouldn't both interfaces be in the routing `main` routing table.

I wonder if somebody can provide a copy of their Troubleshooting tab so that I can see what things are at least supposed to look like.

brianjmurrell, 2020-09-25:
Does `mwan3` expect the network interface configuration to install the default route?

I.e. should I remove the `option defaultroute '0'` from my interface configuration below?

```
config interface 'wan1'
	option ifname 'eth0.3'
	option proto 'pppoe'
...
	option defaultroute '0'
	option ipv6 '1'
	option metric '20'
```

jamesmacwhite, 2020-09-26:
I don't personally use `option defaultroute '0'` on any of my interfaces. I believe mwan3 would expect the default route to be present in the main routing table

aaronjg, 2020-09-26:
Yes, mwan3 expects the default route to be present see ["verify the routing table"](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#prepare_and_the_check_the_default_routing_table_for_the_wan_interfaces) in the doc

> IPv4 and IPv6 being separate was something that was changed in the default config a few versions ago

They actually were always separate, but a few versions ago we added a default config rule for ipv6. Prior to that there was no default rule or ipv6.

brianjmurrell, 2020-09-26:
How is the mwan3 member metric related to the interface metric option?

aaronjg, 2020-09-26:
Member metric applies to the mwan3 policy, and interface metric applies to the default routing table. So if you have wana and wanb, you could have one policy that fails over from wana to wanb for some domains and another that fails over from wanb to wana for other domains

If you want load balancing, then the mwan3 metric should be the same.

ptpt52, 2020-10-26:
I notice this issue, since the current upstream mwan3 in a bad/bug stage for now and it may take a long time to fix

I am sure that my version of mwan3 work great, resolve this issue

https://github.com/x-wrt/packages/commits/master/net/mwan3

feckert, 2020-11-04:
@brianjmurrell For me the failover from wan -> xdsl -> wwan works.
I have noticed the following things that you have already changed?
You have fixed the policy wan0_wan1. 
And the default route of pppoe in the main table you have also recognized.

closed 2020-11-04 by feckert