openwrt/packages #13495: ddns-scripts: DDNS Service provider is not available

peterwillcn, 2020-09-25:
[image]
[image]

snakwu, 2020-09-26:
@feckert
me to！！！

Collected errors:                                                               
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-
scripts_nsupdate:                                                               
 *      bind-client                                                             
 * opkg_install_cmd: Cannot install package ddns-scripts_nsupdate.              
 * check_data_file_clashes: Package ddns-scripts_service wants to install file /
home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/root-x86/usr/sha
re/ddns/services/cnkuai.cn.json                                                 
        But that file is already provided by package  * ddns-scripts_cnkuai_cn  
 * check_data_file_clashes: Package ddns-scripts_service wants to install file /
home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/root-x86/usr/sha
re/ddns/services/freedns.42.pl.json                                             
        But that file is already provided by package  * ddns-scripts_freedns_42_
pl                                                                              
 * check_data_file_clashes: Package ddns-scripts_service wants to install file /
home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/root-x86/usr/sha
re/ddns/services/route53-v1.json                                                
        But that file is already provided by package  * ddns-scripts_route53-v1 
 * opkg_install_cmd: Cannot install package ddns-scripts_service.               
make[2]: *** [package/Makefile:68: package/install] Error 255                   
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                
make[1]: *** [package/Makefile:108: /home/snakwu/openwrt-master/openwrt/staging_
dir/target-x86_64_musl/stamp/.package_install] Error 2                          
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                
make: *** [/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:236: world] E
rror 2

Ansuel, 2020-09-26:
@peterwillcn the app needs to be updated... Can you test the wip pr?

Ansuel, 2020-09-26:
@snakwu we need to fix a bug in the makefile... sorry for the mess

snakwu, 2020-09-26:
> @snakwu we need to fix a bug in the makefile... sorry for the mess

Thank you very much

snakwu, 2020-09-27:
@Ansuel 

Brother, when can I fix it? Thank you

Ansuel, 2020-09-27:
Tomorrow we should push a fix ;)

snakwu, 2020-09-27:
> Tomorrow we should push a fix ;)

ok，；）

ShayBox, 2020-10-02:
How do I apply this fix? A bit new to openwrt, but having the same problem.

Ansuel, 2020-10-02:
if you use the snapshot image just wait for the buildbot to push a new image...

Ansuel, 2020-10-03:
News on this?

peterwillcn, 2020-10-03:
@Ansuel I'll test it soon. Just a moment, please.

peterwillcn, 2020-10-03:
@Ansuel @feckert configuring bug

Ansuel, 2020-10-03:
Can you post your ddns configuration with private data omitted?

peterwillcn, 2020-10-04:
It work,but a little bug.
[image]

feckert, 2020-10-05:
@peterwillcn In my view, this has nothing to do with the last change. That is a different problem.
If it is one at all. The original problem has been fixed

closed 2020-10-05 by feckert

peterwillcn, 2020-10-05:
@feckert I think Package/ ddNs-scripts DEPENDS on :=+ ddNs-scripts-services is not necessary, so it can be set as optional, which will reduce the installation volume.

Ansuel, 2020-10-05:
The complains was exactly about that.... (ddns installs only the core script but not the services)

peterwillcn, 2020-10-05:
Yes, ddNs-scripts-core(ddNs-scripts) and ddNs-scripts-services(optional)

peterwillcn, 2020-10-05:
> The complains was exactly about that.... (ddns installs only the core script but not the services)

Yes, ddNs-scripts-core(ddNs-scripts) and ddNs-scripts-services(optional)