openwrt/packages #13499: mwan: Rule Length

achimbo, 2020-09-26:
Maintainer: @\feckert (find it by checking history of the package Makefile)
Environment: OpenWrt 19.07.2 r10947-65030d81f3, Linksys WRT1900ACS

Description:

I have mwan3	2.8.1-1 up and running, but wenn I enter rules with long names through luci they do not trigger and instead the default rule gets used.

For example willhaben_sso_wan does not work wllhbn_sso_wan does work.

jamesmacwhite, 2020-09-26:
Policies/rule names need to be a maximum of 15 characters. In your example `willhaben_sso_wan` is 17 characters and `wllhbn_sso_wan` is 14, so it will work.

Make sure your rules are 15 characters or less and it will be OK.

achimbo, 2020-09-26:
Yes, I know that short names work. But I believe there should be at least an error message when you enter longer rule names. At least for me it was trial and error.

jamesmacwhite, 2020-09-26:
Fair enough. It is mentioned in the documentation (although this could be made more prominent). Having a warning/message displayed would be beneficial though I would say, so I understand your point.

It could be an enhancement for both mwan3 and luci-app-mwan3 potentially.

achimbo, 2020-09-27:
If you could implement an error message that would be great. I stumbled across this twice, and I am sure it happens to other users as well.

jamesmacwhite, 2020-09-27:
Ultimately, it is up to @feckert, but additional validation in LuCI should be possible.

1. Having a warning displayed in LuCI if a rule name is 15 characters or more is a good suggestion i.e.

[image]

Or even better, could we just stop rules longer than 15 characters being able to be committed if using LuCI?

2. Equally, I don't know if it's worth having mwan3 in console warn about it as well, as not everyone will use LuCI.

I'll leave it open for further discussion.

feckert, 2020-09-28:
We already have a checks in mwan3 for [policies](https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L853) and [rules](https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L981) if the section has more then 15 chars.

In luci-app-mwan3 we have only this check for the [policies](https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/policy.lua#L30). This check is missing for the rules. But this check is only executed if we have already added this to the uci config. I think we should check the value before the config is written.

So we have to extend the lua code with the following lines for policies and rules.

rules:
https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/rule.lua#L63

```
function mwan_rule.create(self, section)
    if #section > 15 then
        self.invalid_cts = true
    else
        TypedSection.create(self, section)
        m.uci:save("mwan3")
        luci.http.redirect(dsp.build_url("admin", "network", "mwan", "rule", section))
    end
end
```

polices:
https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/model/cbi/mwan/policy.lua#L55
```
function mwan_policy.create(self, section)
    if #section > 15 then
        self.invalid_cts = true
    else
        TypedSection.create(self, section)
        m.uci:save("mwan3")
        luci.http.redirect(dsp.build_url("admin", "network", "mwan", "rule", section))
    end
end
```

feckert, 2020-10-14:
Fixed with https://github.com/openwrt/luci/commit/678c0b2d5ba77092ca4c422b6a3c3e3eefe86b01 and
https://github.com/openwrt/luci/commit/8c19a2e1b8f8bb5f28beadd71bab88d8f892d28d

closed 2020-10-14 by feckert