openwrt/packages #13531: gstreamer1-utils builds unusable binaries due to build options

arkanoid87, 2020-09-29:
Maintainer: @flyn-org

> cat /etc/openwrt_release 
> DISTRIB_ID='OpenWrt'
> DISTRIB_RELEASE='19.07.4'
> DISTRIB_REVISION='r11208-ce6496d796'
> DISTRIB_TARGET='ramips/mt76x8'
> DISTRIB_ARCH='mipsel_24kc'
> DISTRIB_DESCRIPTION='OpenWrt 19.07.4 r11208-ce6496d796'
> DISTRIB_TAINTS=''

Description:

This packages list
`"gstreamer1-utils gst1-mod-rtp gst1-mod-rtsp gst1-mod-mpegpsmux"`

should be sufficient to run:
`gst-launch-1.0 rtspsrc location=... ! fakesink`

but gst-launch-1.0 is not finding any element due to disabled registry, moreover disabling cmd parsing leads to the impossibility to load plugins/element manually in /usr/lib/gstreamer-1.0

gst-inspect-1.0 prints:
```
staticelements:  bin: Generic bin
staticelements:  pipeline: Pipeline object

Total count: 1 plugin, 2 features
```

and /usr/lib/gstreamer-1.0/gst-plugin-scanner exits with code 1 without printing anything

how I am supposed to use gst-launch-1.0 with the provided build?

flyn-org, 2020-09-30:
We need to backport this to 19.07. See https://github.com/openwrt/packages/pull/13532.

neheb, 2020-10-01:
Backported.

closed 2020-10-01 by neheb

arkanoid87, 2020-10-02:
now command parsing and registry works

thanks

PR #13532 (2020-09-30): gstreamer1: enable build options necessary for most applications
Commit cb058bf changed gstreamer1's build options, and this broke some
applications.

First, -Doption-parsing=false is intended for very specific embedded
applications rather general packages. Please see:

https://github.com/openwrt/packages/pull/8847
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/388

Second, -Dregistry=false seems to break plugin discovery. Symptoms of
this include broken applications and gst-inspect-1.0 listing zero plugins.

Fixes #13531.

Maintainer: me
Description:
commit a4534f16 (2020-09-30): gstreamer1: enable build options necessary for most applications

Commit eee7ce2b changed gstreamer1's build options, and this broke
some applications.

First, -Doption-parsing=false is intended for very specific embedded
applications rather general packages. Please see:

https://github.com/openwrt/packages/pull/8847
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/388

Second, -Dregistry=false seems to break plugin discovery. Symptoms of
this include broken applications and gst-inspect-1.0 listing zero plugins.

Fixes #13531.

(cherry picked from commit 10aa094d7f7b039b04eec3feab9e [... cut]