openwrt/packages #13537: CircleCI unable to compile Boost

neheb, 2020-10-01:
Boost's python plugin has a weird dependency on at least 2GB of RAM to compile. Which means CircleCI cannot compile it.

GitHub Actions can.

Should we bring back Travis?

ping @openwrt/package-maintainers

aparcar, 2020-10-01:
I'd remove anything but GitHub actions for now. In case GitHub changes their conditions we switch back to CI X. The things are still in the git log anyway.

jefferyto, 2020-10-06:
The CircleCI check also checks if commits are signed off and have a suitable subject line. (I'm not a fan of the "firstname lastname" author check since it assumes all names worldwide are written with a space between the given and family names, but checking that it matches the sign-off name is reasonable.) AFAIK this isn't being done by the GitHub Actions check.

hnyman, 2020-10-06:
There is the separate DCO check for that, I think.
(not sure if that is related to Github or CircleCI)

aparcar, 2020-10-06:
DCO is separate, we could also just migrate the formal checks to GitHub

closed 2022-09-16 by neheb