openwrt/packages #13545: gst1-plugins-bad mpegtsmux/demux missing

arkanoid87, 2020-10-02:
Maintainer: @flyn-org @thess 

```
cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.4'
DISTRIB_REVISION='r11208-ce6496d796'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.4 r11208-ce6496d796'
DISTRIB_TAINTS=''
```
Description:

elements mpegtsmux and mpegtsdemux are missing in build
I guess is is due to commented in Makefile:
https://github.com/openwrt/packages/blob/19866ad4f243ef9ffce6676737e359e6a4c030a7/multimedia/gst1-plugins-bad/Makefile#L383

I'd like to ask if there's an issue in delivering this valid tool into the build.

MpegTS is a valid muxer/demuxer option in an embedded scenario.
In a context where encoding/decoding is impossible/discouraged, mpegts can help in moving and storing live streams

micmac1, 2020-10-04:
It was disabled here: https://github.com/openwrt/packages/commit/e35d46becfdc8026fee3c7cdf5a91049bab0d482

Simply enabling it again isn't enough:

```
Package gst1-mod-mpegtsdemux is missing dependencies for the following libraries:
libgstcodecparsers-1.0.so.0
libgstmpegts-1.0.so.0
```

These libs are part of gst-plugins-bad. So what would need to  be done is add these libs as packages. That means somebody would need to be interested enough to do the work :)

Regards,
Seb

arkanoid87, 2020-10-04:
sadly I don't have the skill set to solve this, but maybe I could help somehow if you would expose some general steps to reproduce the build environment

Thanks

PR #13594 (2020-10-07): gst1-plugins-bad: add back mpegtsdemux & mpegtsmux
These plugins were disabled in commit e35d46b. Add them back. The
required libs are added, too.

Closes #13545

Maintainer: @flyn-org @thess 
Compile tested: ath79 master
Run tested: nope, not bored enough for that :)

Description:
Had a bit of time on my hands and thought I may give #13545 a try. I had initially thought this would be more involved, but the Makefile actually makes it very easy to add libs and plugins. :)

Kind regards,
Seb
commit 55c79de3 (2020-10-07): gst1-plugins-bad: add back mpegtsdemux & mpegtsmux

These plugins were disabled in commit e35d46b. Add them back. The
required libs are added, too.

Closes #13545
closed 2020-10-07 by neheb