openwrt/packages #13553: generate-ipv6-address can not be compiled with ASLR PIE

BKPepe, 2020-10-02:
Maintainer: @PolynomialDivision 
Environment: Environment: aarch64_cortex-a53 (Turris MOX),  OpenWrt master

Description:

I have enabled ASLR PIE 

```
make[3]: Entering directory '/openwrt/build/build_dir/target-aarch64_cortex-a53_musl/generate-ipv6-address-0.1'
ccache_cc -Os -g -Wall   -L/openwrt/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/openwrt/build/staging_dir/target-aarch64_cortex-a53_musl/lib -L/openwrt/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/openwrt/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -fpic -specs=/openwrt/build/include/hardened-ld-pie.specs -znow -zrelro  -o generate-ipv6-address generate-ipv6-address.o 
/openwrt/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: generate-ipv6-address.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `stderr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/openwrt/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: generate-ipv6-address.o(.text.startup+0x1a4): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `stderr'
/openwrt/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:12: generate-ipv6-address] Error 1
make[3]: Leaving directory '/openwrt/build/build_dir/target-aarch64_cortex-a53_musl/generate-ipv6-address-0.1'
make[2]: *** [Makefile:41: /openwrt/build/build_dir/target-aarch64_cortex-a53_musl/generate-ipv6-address-0.1/.built] Error 2
time: package/feeds/packages/generate-ipv6-address/compile#0.22#0.07#0.28
```

PolynomialDivision, 2020-10-03:
Not sure. Maybe the lib was not compiled with `-fpic`?

neheb, 2020-10-07:
-fpic is the problem. Need to compile with -fPIC.

PolynomialDivision, 2020-10-08:
Sry, I need some time to debug this. Is there a time-span until it gets kicked out?

PolynomialDivision, 2020-10-13:
Which Target System is this? I tried using Boradcom BCM47xx/53xx? Seems wrong.

BKPepe, 2021-02-14:
Oh sorry, @PolynomialDivision I haven't noticed that you responded to me. :(

I can compile your added package just fine for Turris 1.x routers (powerpc_8540), Turris Omnia (mvebu/arm_cortex-a9_vfpv3-d16), but it fails for Turris MOX (mvebu/aarch64_cortex-a53). The last one is not in OpenWrt master, yet. But you can easily reproduce it on SoC Marvell Armada 3720, which is using Turris MOX and it is used by others as it is related to aarch64. Some of them are even in OpenWrt like Marvell ESPRESSObin.

BKPepe, 2021-04-19:
Issue persists in the latest `master` and `openwrt-21.02` branches.

neheb, 2021-04-20:
Packages that fail seem to be using autotools or simple Makefile. Wouldn't it be nice for everything to use CMake/meson?

BKPepe, 2021-04-20:
Thanks to @neheb it is going to be fixed by https://github.com/openwrt/packages/pull/15467

closed 2021-04-20 by BKPepe