openwrt/packages #13554: fakepop can not be compiled with enabled ASLR PIE

BKPepe, 2020-10-02:
Maintainer: @Foxtrot
Environment: aarch64_cortex-a53 (Turris MOX), OpenWrt master

Description:

I'm building packages with:
```
CONFIG_PKG_FORTIFY_SOURCE_2=y
CONFIG_PKG_ASLR_PIE_ALL=y
CONFIG_PKG_CC_STACKPROTECTOR_STRONG=y
CONFIG_PKG_RELRO_FULL=y
```

and noticed that fakepop can not be compiled.

```
make[3]: Entering directory '/openwrt-master/build/build_dir/target-aarch64_cortex-a53_musl/fakepop-10'
ccache_cc -D_GNU_SOURCE -Wall -O2 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/include/glib-2.0 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/glib-2.0/include    -c -o fakepop.o fakepop.c
cc1: note: someone does not honour COPTS correctly, passed 0 times
ccache_cc -D_GNU_SOURCE -Wall -O2 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/include/glib-2.0 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/glib-2.0/include    -c -o header.o header.c
cc1: note: someone does not honour COPTS correctly, passed 0 times
ccache_cc -D_GNU_SOURCE -Wall -O2 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/include/glib-2.0 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/glib-2.0/include    -c -o id.o id.c
cc1: note: someone does not honour COPTS correctly, passed 0 times
ccache_cc -D_GNU_SOURCE -Wall -O2 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/include/glib-2.0 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/glib-2.0/include    -c -o msg.o msg.c
cc1: note: someone does not honour COPTS correctly, passed 0 times
ccache_cc -D_GNU_SOURCE -Wall -O2 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/include/glib-2.0 -I/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/glib-2.0/include    -c -o retrieve.o retrieve.c
cc1: note: someone does not honour COPTS correctly, passed 0 times
ccache_cc -L/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/lib -L/openwrt-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/openwrt-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -L/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-full/lib -Wl,-rpath-link=/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-full/lib -L/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-full/lib -Wl,-rpath-link=/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-full/lib -fpic -specs=/openwrt-master/build/include/hardened-ld-pie.specs -znow -zrelro   fakepop.o header.o id.o msg.o retrieve.o  -L/openwrt-master/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -lglib-2.0 -lintl  -o fakepop
/openwrt-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: fakepop.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `stdout' which may bind externally can not be used when making a shared object; recompile with -fPIC
/openwrt-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: fakepop.o(.text+0x1b4): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `stdout'
/openwrt-master/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[3]: *** [<builtin>: fakepop] Error 1
make[3]: Leaving directory '/openwrt-master/build/build_dir/target-aarch64_cortex-a53_musl/fakepop-10'
make[2]: *** [Makefile:45: /openwrt-master/build/build_dir/target-aarch64_cortex-a53_musl/fakepop-10/.built] Error 2
time: package/feeds/packages/fakepop/compile#0.66#0.27#0.91
```

In past, I created a few similar issues about ASLR PIE, maybe you can find in one of them some help.

neheb, 2020-10-05:
I really think we need a buildbot for this. ping @aparcar

neheb, 2020-10-05:
ping @BKPepe is this a bug with ASLR or with cccache not passing arguments properlu?

BKPepe, 2020-10-07:
> ping @BKPepe is this a bug with ASLR or with cccache not passing arguments properlu?

This is bug with ASLR. This is how it can be reproduced with GCC 8.4.0:

git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a

In `make menuconfig` select:

```
- Target System (Marvell EBU Armada)
- Subtarget (Marvell Armada 3700LP (ARM64)
- Target Profile e.g. Marvell ESPRESSObin Non-eMMC

```
Global build settings:
```
User space ASLR PIE (all) = CONFIG_PKG_ASLR_PIE_ALL=y
User space Stack-Smashing Protection (Strong) = CONFIG_PKG_CC_STACKPROTECTOR_STRONG=y
User space Stack-Smashing Protection (Strong) =  KERNEL_CC_STACKPROTECTOR_STRONG=y
Enable buffer-overflows detection (Aggressive)  = CONFIG_PKG_FORTIFY_SOURCE_2=y
Enable RELRO protection (full) = CONFIG_PKG_RELRO_FULL=y
```

Then: 
```
make diffconfig

make tools/compile -j8
make toolchain/compile -j8
make target/compile -j8
```

Afterwards, let's try to compile `fakepop`, but you need to enable it in `make menuconfig`

This is how it ends:
```
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -fmacro-prefix-map=/home/pepe/tryingissue13554/build_dir/target-aarch64_cortex-a53_musl/fakepop-10=fakepop-10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/include -I/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include/fortify -I/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -fmacro-prefix-map=/home/pepe/tryingissue13554/build_dir/target-aarch64_cortex-a53_musl/fakepop-10=fakepop-10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/include -I/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include/fortify -I/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include " LDFLAGS="-L/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/lib -L/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -L/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/lib -L/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/lib -fpic -specs=/home/pepe/tryingissue13554/include/hardened-ld-pie.specs -znow -zrelro " make -j1 -C /home/pepe/tryingissue13554/build_dir/target-aarch64_cortex-a53_musl/fakepop-10/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/include -I/home/pepe/tryingissue13554/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/include -fmacro-prefix-map=/home/pepe/tryingissue13554/build_dir/target-aarch64_cortex-a53_musl/fakepop-10=fakepop-10 -Wformat -Werror=format-security -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[3]: Entering directory '/home/pepe/tryingissue13554/build_dir/target-aarch64_cortex-a53_musl/fakepop-10'
[... cut ...]
/home/pepe/tryingissue13554/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[3]: *** [<builtin>: fakepop] Error 1
make[3]: Leaving directory '/home/pepe/tryingissue13554/build_dir/target-aarch64_cortex-a53_musl/fakepop-10'
make[2]: *** [Makefile:45: /home/pepe/tryingissue13554/build_dir/target-aarch64_cortex-a53_musl/fakepop-10/.built] Error 2
```

aparcar, 2020-10-07:
Please ping me if it needs another SDK to test this automatically. I can't add anything ontopic.

PR #16420 (2021-08-24): fakepop: fix compilation with PKG_ASLR
Maintainer: @foxtrot 
Compile tested: ath79

Fixes: #13554
closed 2021-08-24 by neheb