openwrt/packages #13555: tinc: Adding package for version 1.1

flosoft, 2020-10-02:
Maintainer: @zioproto
Environment: Turris OS 5.1.1 (based on OpenWrt 19.07)

Description:
Would it be possible to add support for tinc 1.1? I've come across a fork [here](https://github.com/maddie/openwrt-tinc-1.1), but not quite sure how to get it compiled for the Turris Omnia router.
It would be great if there's a separate package for tinc 1.1 available in the repository.

zioproto, 2020-10-02:
Hello,

according to the main project website:
https://www.tinc-vpn.org/

tinc 1.0.36 is the stable supported version.

Looking at the 1.1 tree:
https://www.tinc-vpn.org/git/browse?p=tinc;a=shortlog;h=refs/heads/1.1

It looks like there were very few commits in 2019 and 2020

@gsliepen do you have any plan for a stable release of tinc 1.1 ?

There are many forks on github, if you do a search for:
`tinc "PKG_VERSION" "pre"`

You will find many many forks.

So the real problem now is not creating the package for 1.1 but maintaining it.

It is since 2016 that I am looking for a successor, because I am not actively using tinc anymore, so I dont run the infrastructure to test the changes.

@flosoft Are you interested ? I could review your patches if you want to submit a patch of a official tinc-1.1 openwrt package

thanks

flosoft, 2020-10-03:
I'm happy to help where I can. I use tinc regularly, so happy to help test it. I'm just not familiar as to how to compile the package in the first place to then test it.
It looks like @maddie is has a well maintained fork for OpenWrt that could be a good starting off point: https://github.com/maddie/openwrt-tinc-1.1

maddie, 2020-10-09:
I have a build script for this, just download the OpenWRT SDK for your device, decompress it, cd into it then run:

```
#!/bin/sh

./scripts/feeds update -a
rm -rf feeds/package/net/tinc
git clone https://github.com/maddie/openwrt-tinc-1.1 package/packages/tinc
./scripts/feeds install tinc
make defconfig
./scripts/feeds install zlib-dev liblzo libopenssl
make package/tinc/compile V=sc
```

The resulting ipk will be in `./bin/packages` directory.

And IPK for Turris Omnia [can be found here.](https://github.com/maddie/openwrt-tinc-1.1/releases/download/19.07.0/tinc_1.1pre17_arm_cortex-a9_vfpv3-d16.ipk)

zioproto, 2020-10-12:
@maddie would you be interested in taking ownership of tinc in openwrt/packages ?

maddie, 2020-10-13:
@zioproto Sorry, I'm afraid that I'm not able to commit to maintaining this package, and I don't really know how the whole system works. The steps above are a result of trial-and-error I've been through for several hours 😂

ErwanMAS, 2021-06-09:
FYI version 1.1 is the  version in 21.02 and develppement

closed 2025-01-06 by zioproto