openwrt/packages #13568: https-dns-proxy: provide an option for manual dns configuration

bjmi, 2020-10-04:
Maintainer: @stangri
Environment: OpenWrt 19.07.4

Description:
Please provide an option for https-dns-proxy that if enabled effects
* leaves current dnsmasq configuration untouched
* skips creation of a backup in dhcp section
* skips restart of dnsmasq

i.e. just start the configured https-proxy-instances and let the user configure dnsmasq server entries by himself.

The option could be named "Dnsmasq auto-configuration" and is checked by default.

stangri, 2020-10-04:
Check the configuration settings in the Readme.

bjmi, 2020-10-05:
Usage of **-** works. Thank you.
```
config main 'config'
	option update_dnsmasq_config '-'
```
It wasn't obvious what the opposite of **\*** is.
Maybe luci-app-https-dns-proxy should provide this option too.

closed 2020-10-05 by bjmi

stangri, 2020-10-05:
> Maybe luci-app-https-dns-proxy should provide this option too.

Yeah, I was thinking of how to best implement it. With luci app I can only show `*`, `-` and then parse dnsmasq for its configs and list available (would be just [0] for most people). CLI/uci tho supports settings like '025' to enable automatic dnsmasq update for configs [0], [2] and [5] and there are no ready controls for that in luci API. I haven't looked at the new javascript API, maybe it's possible there.

stangri, 2020-10-06:
@bjmi can you check out https://github.com/stangri/repo.openwrt.melmac.net/blob/master/luci-app-https-dns-proxy_git-20.267.56601-1da9df8_all.ipk and provide feedback?

bjmi, 2020-10-08:
All 3 configurations were tested.
* Update all configs
* Update dhcp.@dnsmasq[0] config
* Do not update configs

The first two options update dnsmasq config on start and revert config on stop.

Note: I only have one dnsmasq configuration i.e. second option couldn't be tested with multiple entries.
Thanks again.

stangri, 2020-10-09:
> All 3 configurations were tested.
> 
> * Update all configs
> * Update dhcp.@dnsmasq[0] config
> * Do not update configs
> 
> The first two options update dnsmasq config on start and revert config on stop.

Great, thanks for testing this. If you have any other suggestions for the WebUI improvements, feel free to create an issue.

> Btw: Quad 9 Recommended have ipv6 servers too.
> 9.9.9.9
> 149.112.112.112
> 2620:fe::fe
> 2620:fe::9

Do you want to create your own PR or would you be happy if I incorporate these changes and create the PR myself?

bjmi, 2020-10-09:
> Great, thanks for testing this. If you have any other suggestions for the WebUI improvements, feel free to create an issue.

You are welcome. 

> > Btw: Quad 9 Recommended have ipv6 servers too.
> > 9.9.9.9
> > 149.112.112.112
> > 2620:fe::fe
> > 2620:fe::9
> 
> Do you want to create your own PR or would you be happy if I incorporate these changes and create the PR myself?

The overhead is far less if you create the PR by yourself.
Quad9 References - https://www.quad9.net/faq/#Is_there_IPv6_support_for_Quad9

stangri, 2020-10-10:
@bjmi can you please test https://repo.openwrt.melmac.net/luci-app-https-dns-proxy_git-20.267.56601-1da9df8_all.ipk for Quad9 bootstrap IPv6 servers.

bjmi, 2020-10-12:
> @bjmi can you please test https://repo.openwrt.melmac.net/luci-app-https-dns-proxy_git-20.267.56601-1da9df8_all.ipk for Quad9 bootstrap IPv6 servers.

Resolver _Quad 9 (Recommended), Quad 9 (Secured with ECS Support), Quad 9 (Secured) and Quad 9 (Unsecured)_ were started and DNS resolution using `dig @routerip -p resolverport anydomain.com` was successful.

IPs and resolver url  from `/usr/lib/lua/luci/https-dns-proxy/providers/net.quad9.*.lua` were double-checked with https://www.quad9.net/doh-quad9-dns-servers/#UsingDoHwithQuad9DNSServers-AdditionalInformation

`ps` shows new IPv6 addresses too
```
  PID USER       VSZ STAT COMMAND
32338 nobody    2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5099 -b 9.9.9.9,149.112.112.112,2620:fe::fe,2620:fe::9 -r https://dns.quad9.net/dns-query -u nobody -g nogroup
32339 nobody    2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5098 -b 9.9.9.11,149.112.112.11,2620:fe::11,2620:fe::fe:11 -r https://dns11.quad9.net/dns-query -u nobody -g nogroup -4
32340 nobody    2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5097 -b 9.9.9.9,149.112.112.9,2620:fe::fe,2620:fe::9 -r https://dns9.quad9.net/dns-query -u nobody -g nogroup -4
32559 root      2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5096 -b 9.9.9.10,149.112.112.10,2620:fe::10, 2620:fe::fe:10 -r https://dns10.quad9.net/dns-query -u nobody -g nogroup -4
```
(Strange: Quad 9 unsecured was started as root.)

stangri, 2020-10-12:
@bjmi thank you for thorough testing. I'm concerned that the first entry is ran without `-4` and the last one is running from root. I cannot reproduce this.

bjmi, 2020-10-13:
> @bjmi thank you for thorough testing. I'm concerned that the first entry is ran without `-4` and the last one is running from root. I cannot reproduce this.

There is a space in 2620:fe::10, 2620:fe::fe:10

stangri, 2020-10-13:
OMG, good catch, I would have never found it!

stangri, 2020-10-13:
> There is a space in `2620:fe::10, 2620:fe::fe:10`

OMG, good catch, I would have never found this. Same link for the IPK if you can test it again.

bjmi, 2020-10-27:
> OMG, good catch, I would have never found this. Same link for the IPK if you can test it again.

```
32250 nobody    2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5098 -b 9.9.9.10,149.112.112.10,2620:fe::10,2620:fe::fe:10 -r https://dns10.quad9.net/dns-query -u nobody -g nogroup -4
32251 nobody    2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5097 -b 9.9.9.9,149.112.112.9,2620:fe::fe,2620:fe::9 -r https://dns9.quad9.net/dns-query -u nobody -g nogroup -4
32252 nobody    2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5096 -b 9.9.9.9,149.112.112.112,2620:fe::fe,2620:fe::9 -r https://dns.quad9.net/dns-query -u nobody -g nogroup -4
32356 nobody    2168 S    /usr/sbin/https-dns-proxy -a 0.0.0.0 -p 5099 -b 9.9.9.11,149.112.112.11,2620:fe::11,2620:fe::fe:11 -r https://dns11.quad9.net/dns-query -u nobody -g nogroup -4
```
Looks good to me now. Thank you. Although I don't really know what _-4_ does :)

stangri, 2020-10-29:
Thanks, I'll try to submit PR/merge asap. You should get a ping for the mention.