openwrt/packages #13585: [Meta] Hacktoberfest 2020

ja-pa, 2020-10-06:
Will packages repo participate in this year Hacktoberfest?

This could be done by creating hacktoberfest-accepted label for specific PR or by creating hacktoberfest topic in the repo.

More details are here 
https://hacktoberfest.digitalocean.com/details#details

aparcar, 2020-10-06:
I think participating is fine but the maintainers will likely face some annoyances. See https://drewdevault.com/2020/10/01/Spamtoberfest.html

I'm unsure what people can distribute here, are there any "good first issue" labels used?

aparcar, 2020-10-06:
@ja-pa if you in particular want the shirt I'm happy to review and tag your (valid) pull requests

ja-pa, 2020-10-06:
> @ja-pa if you in particular want the shirt I'm happy to review and tag your (valid) pull requests

Yes, I want a t-shirt :-) , but I also think that is a good idea which should be supported.

aparcar, 2020-10-06:
I can't change the labels for openwrt/packages - @neheb do you have admin status here?

neheb, 2020-10-06:
I do not.

It seems there's already a hacktoberfest-accepted label here.

aparcar, 2020-10-06:
I added it to specific PRs of @ja-pa, however not adding it to the the repo labels means that it's not "promoted" via DO.

blocktrron, 2020-10-07:
I've added the topic to the packages repository.

The merge strategy for packages should therefor be enough to count towards the goal. If this is somewhat wrong / not the case feel free to reopen.

closed 2020-10-07 by blocktrron