openwrt/packages #13589: Runtime test ideas/enhancements

jefferyto, 2020-10-06:
Maintainer: @aparcar (also ping @neheb)
Environment: N/A

Description:
(These are ideas that occurred to me from reading #13449.)

Instead of having every package include a `test.sh`, I'd like to suggest having a default runtime test that:

* Checks to see if the package contains at least one executable
* Tries running the executable with various versions of the `--version` command line option, e.g. `foo --version`, `foo -version`, `foo version`
* If any attempt exits with a success and the version number is found in the output, then the test is considered successful

It would be great if there is a variable that can be added to the package Makefile to disable this default runtime test.

If a custom runtime test is very simple, it would also be great if the test code can be written directly in the package Makefile, i.e. assigned to a Makefile variable (perhaps the same as the previous variable).

BKPepe, 2020-10-16:
@aparcar Any response would be highly appreciated. This is here for 10 days. It does not look good without a reply.

aparcar, 2020-10-16:
Hi, thanks for pinging me, this issues got lost in my notifications.

> Checks to see if the package contains at least one executable

That sounds great, could you share you implementations ideas? We can unpack the package in a temporary folder and scan for executables. How do we deal with a package like `vim-docs` and how with libraries? Are package like these rare enough to require maintainers to adopt a Makefile?

> Tries running the executable with various versions of the --version command line option, e.g. foo --version, foo -version, foo version

I had a similar idea but chose the `test.sh` way because it's more conscious to enable tests, rather than having them enabled by default. However, as people seem to like runtime tests, let's enable this version checking command.

> It would be great if there is a variable that can be added to the package Makefile to disable this default runtime test.

Sure let's do this. As written before I meant to have it optional but we can also make it an opt-out.

> If a custom runtime test is very simple, it would also be great if the test code can be written directly in the package Makefile, i.e. assigned to a Makefile variable (perhaps the same as the previous variable).

We could introduce a variable `PKG_TEST` which can also be added on a *subpackage* base:
* Empty or `default`: Run automated version checks
* `no` (or `0` or `none` or ...?) disables tests
* `script` run a `test.sh` script imitating the current behaviour
* Anything else is run as a script?

I'm a bit afraid of allowing actual scripts in the Makefiles, people may get fancy and bloat Makefiles?

aparcar, 2020-10-26:
> It does not look good without a reply.

jefferyto, 2020-10-27:
> > It does not look good without a reply.

Is this is directed at me?

> We can unpack the package in a temporary folder and scan for executables.

I think looking for executables in `/bin`, `/sbin`, `/usr/bin` and `/usr/sbin` should be enough.

> How do we deal with a package like `vim-docs` and how with libraries?

I assume libraries exist on the device to support programs, so testing the programs would be enough.

> I had a similar idea but chose the `test.sh` way because it's more conscious to enable tests, rather than having them enabled by default.

Thinking about this a bit more, I agree that having it be opt-in is better, though I still would like to have a default test routine that can be enabled (like setting `PKG_INSTALL:=1` to get the default package install routine).

> I'm a bit afraid of allowing actual scripts in the Makefiles, people may get fancy and bloat Makefiles?

Makefiles already contain scripts: `Build/Compile`, `Package/xxx/install`, etc.