openwrt/packages #13591: Network pppoe error

9000000, 2020-10-07:
```
08:49:32 2020 daemon.notice netifd: Interface 'wan' is setting up now
Wed Oct  7 08:49:32 2020 kern.info kernel: [86737.410108] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
Wed Oct  7 08:49:32 2020 daemon.err insmod: module is already loaded - slhc
Wed Oct  7 08:49:32 2020 daemon.err insmod: module is already loaded - ppp_generic
Wed Oct  7 08:49:32 2020 daemon.err insmod: module is already loaded - pppox
Wed Oct  7 08:49:32 2020 daemon.err insmod: module is already loaded - pppoe
Wed Oct  7 08:49:32 2020 daemon.info pppd[29412]: Plugin rp-pppoe.so loaded.
Wed Oct  7 08:49:32 2020 daemon.info pppd[29412]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Wed Oct  7 08:49:32 2020 daemon.notice pppd[29412]: pppd 2.4.8 started by root, uid 0
Wed Oct  7 08:49:45 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 0c:3e:9f:18:fb:c6
Wed Oct  7 08:49:45 2020 daemon.info hostapd: wlan1: STA 0c:3e:9f:18:fb:c6 IEEE 802.11: disassociated
Wed Oct  7 08:49:46 2020 daemon.info hostapd: wlan1: STA 0c:3e:9f:18:fb:c6 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Oct  7 08:49:47 2020 daemon.info hostapd: wlan1: STA 0c:3e:9f:18:fb:c6 IEEE 802.11: authenticated
Wed Oct  7 08:49:47 2020 daemon.info hostapd: wlan1: STA 0c:3e:9f:18:fb:c6 IEEE 802.11: associated (aid 2)
Wed Oct  7 08:49:47 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 0c:3e:9f:18:fb:c6
Wed Oct  7 08:49:47 2020 daemon.info hostapd: wlan1: STA 0c:3e:9f:18:fb:c6 WPA: pairwise key handshake completed (RSN)
Wed Oct  7 08:49:47 2020 daemon.info dnsmasq-dhcp[11437]: DHCPREQUEST(br-lan) 192.168.7.182 0c:3e:9f:18:fb:c6
Wed Oct  7 08:49:47 2020 daemon.info dnsmasq-dhcp[11437]: DHCPACK(br-lan) 192.168.7.182 0c:3e:9f:18:fb:c6 Luu-Kieu
Wed Oct  7 08:49:47 2020 daemon.warn pppd[29412]: Timeout waiting for PADO packets
Wed Oct  7 08:49:47 2020 daemon.err pppd[29412]: Unable to complete PPPoE Discovery
Wed Oct  7 08:49:47 2020 daemon.info pppd[29412]: Exit.
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' is now down
Wed Oct  7 08:49:47 2020 kern.info kernel: [86752.720280] mt7530 mdio-bus:1f wan: Link is Down
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' is disabled
Wed Oct  7 08:49:47 2020 kern.info kernel: [86752.736710] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
Wed Oct  7 08:49:47 2020 kern.info kernel: [86752.744055] 8021q: adding VLAN 0 to HW filter on device wan
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' is enabled
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' is setting up now
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Network device 'wan' link is down
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' is now down
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' is disabled
Wed Oct  7 08:49:47 2020 kern.info kernel: [86752.911380] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
Wed Oct  7 08:49:47 2020 kern.info kernel: [86752.918721] 8021q: adding VLAN 0 to HW filter on device wan
Wed Oct  7 08:49:47 2020 daemon.notice netifd: Interface 'wan' is enabled
Wed Oct  7 08:49:49 2020 user.notice mwan3-hotplug[29551]: set_user_iface_rules: could not find device corresponding to iface wan
Wed Oct  7 08:49:49 2020 user.notice mwan3-hotplug[29551]: Execute ifdown event on interface wan (unknown)
Wed Oct  7 08:49:49 2020 daemon.info dnsmasq-dhcp[11437]: DHCPREQUEST(br-lan) 192.168.7.182 0c:3e:9f:18:fb:c6
Wed Oct  7 08:49:49 2020 daemon.info dnsmasq-dhcp[11437]: DHCPACK(br-lan) 192.168.7.182 0c:3e:9f:18:fb:c6 Luu-Kieu
Wed Oct  7 08:49:49 2020 user.info mwan3track[21843]: Detect ifdown event on interface wan (pppoe-wan)
Wed Oct  7 08:49:52 2020 daemon.notice netifd: Network device 'wan' link is up
Wed Oct  7 08:49:52 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Wed Oct  7 08:49:52 2020 daemon.notice netifd: Interface 'wan' is setting up now
Wed Oct  7 08:49:52 2020 kern.info kernel: [86757.025994] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
Wed Oct  7 08:49:52 2020 daemon.err insmod: module is already loaded - slhc
Wed Oct  7 08:49:52 2020 daemon.err insmod: module is already loaded - ppp_generic
Wed Oct  7 08:49:52 2020 daemon.err insmod: module is already loaded - pppox
Wed Oct  7 08:49:52 2020 daemon.err insmod: module is already loaded - pppoe
Wed Oct  7 08:49:52 2020 daemon.info pppd[30032]: Plugin rp-pppoe.so loaded.
Wed Oct  7 08:49:52 2020 daemon.info pppd[30032]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Wed Oct  7 08:49:52 2020 daemon.notice pppd[30032]: pppd 2.4.8 started by root, uid 0
Wed Oct  7 08:49:52 2020 daemon.info pppd[30032]: PPP session is 1308
Wed Oct  7 08:49:52 2020 daemon.warn pppd[30032]: Connected to 82:ac:ac:ee:56:b0 via interface wan
Wed Oct  7 08:49:52 2020 kern.info kernel: [86757.228085] pppoe-wan: renamed from ppp0
Wed Oct  7 08:49:52 2020 daemon.info pppd[30032]: Renamed interface ppp0 to pppoe-wan
Wed Oct  7 08:49:52 2020 daemon.info pppd[30032]: Using interface pppoe-wan
Wed Oct  7 08:49:52 2020 daemon.notice pppd[30032]: Connect: pppoe-wan <--> wan
Wed Oct  7 08:49:55 2020 daemon.notice pppd[30032]: Modem hangup
Wed Oct  7 08:49:55 2020 daemon.notice pppd[30032]: Connection terminated.
Wed Oct  7 08:49:55 2020 daemon.info pppd[30032]: Sent PADT
Wed Oct  7 08:49:55 2020 daemon.info pppd[30032]: Exit.
Wed Oct  7 08:49:55 2020 daemon.notice netifd: Interface 'wan' is now down
Wed Oct  7 08:49:55 2020 kern.info kernel: [86760.391201] mt7530 mdio-bus:1f wan: Link is Down
Wed Oct  7 08:49:55 2020 daemon.notice netifd: Interface 'wan' is disabled
Wed Oct  7 08:49:55 2020 kern.info kernel: [86760.407797] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
Wed Oct  7 08:49:55 2020 kern.info kernel: [86760.415117] 8021q: adding VLAN 0 to HW filter on device wan
Wed Oct  7 08:49:55 2020 daemon.notice netifd: Interface 'wan' is enabled
Wed Oct  7 08:49:55 2020 daemon.notice netifd: Interface 'wan' is setting up now
Wed Oct  7 08:49:55 2020 daemon.notice netifd: Network device 'wan' link is down
[... 103 lines cut ...]
Wed Oct  7 08:50:58 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Wed Oct  7 08:50:58 2020 daemon.notice netifd: Interface 'wan' is setting up now
Wed Oct  7 08:50:58 2020 kern.info kernel: [86823.393657] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
Wed Oct  7 08:50:58 2020 daemon.err insmod: module is already loaded - slhc
Wed Oct  7 08:50:58 2020 daemon.err insmod: module is already loaded - ppp_generic
Wed Oct  7 08:50:58 2020 daemon.err insmod: module is already loaded - pppox
Wed Oct  7 08:50:58 2020 daemon.err insmod: module is already loaded - pppoe
Wed Oct  7 08:50:58 2020 daemon.info pppd[32575]: Plugin rp-pppoe.so loaded.
Wed Oct  7 08:50:58 2020 daemon.info pppd[32575]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Wed Oct  7 08:50:58 2020 daemon.notice pppd[32575]: pppd 2.4.8 started by root, uid 0
Wed Oct  7 08:51:13 2020 daemon.warn pppd[32575]: Timeout waiting for PADO packets
Wed Oct  7 08:51:13 2020 daemon.err pppd[32575]: Unable to complete PPPoE Discovery
Wed Oct  7 08:51:13 2020 daemon.info pppd[32575]: Exit.
Wed Oct  7 08:51:13 2020 daemon.notice netifd: Interface 'wan' is now down
Wed Oct  7 08:51:13 2020 kern.info kernel: [86838.729320] mt7530 mdio-bus:1f wan: Link is Down
Wed Oct  7 08:51:13 2020 daemon.notice netifd: Interface 'wan' is disabled
Wed Oct  7 08:51:13 2020 kern.info kernel: [86838.755591] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
Wed Oct  7 08:51:13 2020 kern.info kernel: [86838.762991] 8021q: adding VLAN 0 to HW filter on device wan
Wed Oct  7 08:51:13 2020 daemon.notice netifd: Interface 'wan' is enabled
Wed Oct  7 08:51:13 2020 daemon.notice netifd: Interface 'wan' is setting up now
Wed Oct  7 08:51:13 2020 daemon.notice netifd: Network device 'wan' link is down
Wed Oct  7 08:51:13 2020 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed Oct  7 08:51:13 2020 daemon.notice netifd: Interface 'wan' is now down
Wed Oct  7 08:51:13 2020 daemon.notice netif
```

9000000, 2020-10-07:
My device is Xiaomi 3 pro currently on the latest build

feckert, 2020-10-07:
This package is maitained in https://github.com/openwrt/openwrt/tree/master/package/network/services/ppp and not in the package feed. Did that even work with your setup? Did you configure it correctly? Also which software version do you use.

neheb, 2020-10-07:
This is for https://bugs.openwrt.org

9000000, 2020-10-10:
I always use the latest test builds, if I use the 19.07 version there will be no bugs 

```
Model
Xiaomi Mi Router 3 Pro
Architecture
MediaTek MT7621 ver:1 eco:3
Firmware Version
OpenWrt SNAPSHOT r14660-5423d9d27e / LuCI Master git-20.280.41447-12e50de
Kernel Version
5.4.70
```

```
daemon.notice pppd[8405]: pppd 2.4.8 started by root, uid 0
Sat Oct 10 23:42:25 2020 daemon.err pppd[8405]: error sending pppoe packet: No buffer space available
Sat Oct 10 23:42:29 2020 kern.info kernel: [  246.652693] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
Sat Oct 10 23:42:29 2020 daemon.notice netifd: Network device 'wan' link is up
Sat Oct 10 23:42:29 2020 daemon.notice netifd:
```

9000000, 2020-10-11:
I have read a lot of information regarding this error

https://forum.openwrt.org/t/archer-c60-wan-disconnects/68896

https://forum.openwrt.org/t/pppoe-disconnects-all-the-time/71165/23

lucize, 2020-10-12:
lately I'm having pppoe problems on MT7621, once the disconnect is starting it will happen very often until I reboot the router (Ubiquiti EdgeRouter X) I'll try to switch to another chipset, I think this started happening after hw-offload being corrected for DSA switch, with hw-offload enabled it is happening more often or I think so, for now not sure if is the provider, it had 99.99% uptime previously

```
Sun Oct 11 15:17:38 2020 daemon.info pppd[2776]: No response to 5 echo-requests
Sun Oct 11 15:17:38 2020 daemon.notice pppd[2776]: Serial link appears to be disconnected.
Sun Oct 11 15:17:38 2020 daemon.info pppd[2776]: Connect time 3806.5 minutes.
Sun Oct 11 15:17:38 2020 daemon.info pppd[2776]: Sent 2197345008 bytes, received 2360722919 bytes.
[...]
Sun Oct 11 15:35:42 2020 daemon.notice pppd[14100]: Connection terminated.
Sun Oct 11 15:35:42 2020 daemon.info pppd[14100]: Sent PADT
Sun Oct 11 15:35:42 2020 daemon.info pppd[14100]: Exit.
Sun Oct 11 15:35:42 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Oct 11 15:35:42 2020 kern.info kernel: [229502.403026] mt7530 mdio-bus:1f eth0: Link is Down
Sun Oct 11 15:35:42 2020 daemon.notice netifd: Interface 'wan' is disabled
[...]
Sun Oct 11 15:35:49 2020 daemon.info pppd[15214]: Using interface pppoe-wan
Sun Oct 11 15:35:49 2020 daemon.notice pppd[15214]: Connect: pppoe-wan <--> eth0
Sun Oct 11 15:35:49 2020 daemon.debug dnsmasq[13446]: listening on eth0(#3): fe80::46d9:e7ff:fe07:2ed4%eth0 port 53
Sun Oct 11 15:36:19 2020 daemon.warn pppd[15214]: LCP: timeout sending Config-Requests
Sun Oct 11 15:36:19 2020 daemon.notice pppd[15214]: Connection terminated.
Sun Oct 11 15:36:19 2020 daemon.info pppd[15214]: Sent PADT
Sun Oct 11 15:36:19 2020 daemon.notice pppd[15214]: Modem hangup
Sun Oct 11 15:36:19 2020 daemon.info pppd[15214]: Exit.
Sun Oct 11 15:36:19 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Oct 11 15:36:19 2020 kern.info kernel: [229538.795970] mt7530 mdio-bus:1f eth0: Link is Down

```

lucize, 2020-10-25:
I have uptime of 8 days with r14715-e17529a3e3, not sure what was the problem, maybe ISP

feckert, 2020-11-04:
This is a core problem and is not related to the package feed.
Please open a bug on https://bugs.openwrt.org/

closed 2020-11-04 by feckert