openwrt/packages #13608: luci-app-statistics	is not creating graphs

sander816, 2020-10-08:
# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: archer C7, LuCI openwrt-19.07 branch (git-20.247.75781-0d0ab01) / OpenWrt 19.07.4 r11208-ce6496d796

Description:
luci-app-statistics	is not creating graphs
Collectd is complaing about wrong directory:
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory

```
Thu Oct  8 22:09:34 2020 daemon.err collectd[7867]: Initialization complete, entering read-loop.
Thu Oct  8 22:09:36 2020 daemon.info dnsmasq[5419]: read /etc/hosts - 4 addresses
Thu Oct  8 22:09:36 2020 daemon.info dnsmasq[5419]: read /tmp/hosts/odhcpd - 2 addresses
Thu Oct  8 22:09:36 2020 daemon.info dnsmasq[5419]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Oct  8 22:09:36 2020 daemon.info dnsmasq-dhcp[5419]: read /etc/ethers - 0 addresses
Thu Oct  8 22:24:35 2020 daemon.err collectd[7867]: Exiting normally.
Thu Oct  8 22:24:35 2020 daemon.err collectd[7867]: collectd: Stopping 2 read threads.
Thu Oct  8 22:24:35 2020 daemon.err collectd[7867]: collectd: Stopping 5 write threads.
Thu Oct  8 22:24:35 2020 daemon.err collectd[7867]: rrdtool plugin: Shutting down the queue thread.

Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory

Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: plugin_load: plugin "iwinfo" successfully loaded.
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: plugin_load: plugin "memory" successfully loaded.
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: plugin_load: plugin "cpu" successfully loaded.
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: plugin_load: plugin "rrdtool" successfully loaded.
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: rrdtool plugin: RRASingle = true: creating only AVERAGE RRAs
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: plugin_load: plugin "interface" successfully loaded.
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: plugin_load: plugin "load" successfully loaded.
Thu Oct  8 22:24:35 2020 daemon.err collectd[8157]: Initialization complete, entering read-loop.```

feckert, 2020-10-10:
To fix this create this directory by hand. Then collectd should work. The include directory is set in the configuration https://github.com/openwrt/luci/blob/53c6c7f905b516d19b1e719b28216e027249dde7/applications/luci-app-statistics/root/etc/config/luci_statistics#L5. To make this work without user interaction we should create this directory during [service](https://github.com/openwrt/luci/blob/master/applications/luci-app-statistics/root/etc/init.d/luci_statistics#L13) start. You could add the following line `mkdir -p /etc/collectd/conf.d` to the init script. But for this to work even if someone changes the configuration you have to read the configuration first to create the directory depending on the configuration.

PolynomialDivision, 2021-02-11:
>  luci-app-statistics is not creating graphs

I have this error, too, but I have graphs using OpenWrt master.

PolynomialDivision, 2021-02-11:
ping @hnyman

hnyman, 2021-02-11:
> configfile: stat (/etc/collectd/conf.d) failed: No such file or directory

That error is harmless, and graphs work quite ok without that directory.

closed 2021-02-11 by hnyman