openwrt/packages #13616: [netdata] upgrade to source release 1.25.0

n8v8R, 2020-10-09:
Maintainer: @BKPepe @diizzyy 

Description:

[source](https://github.com/netdata/netdata/tags) has release 1.25.0 with plenty of bug fixes. Would be appreciate to be available here as well.

BKPepe, 2020-10-10:
Is there anything interested in why you would like to have the latest Netdata 1.25.0 release? Any pull requests are welcome.

n8v8R, 2020-10-10:
If you are not interested to keep the package on par with source development then just say so and close the issue, preferably stating your reasoning as opposed to question the motivation of keeping a package on par with its source, notwithstanding asking of a PR.

BKPepe, 2020-10-10:
Didn't I ask you for a simple question? What is the most interesting feature for you in the latest releases? And now you trying to offend me. Unfortunately, this is not how it works. If you want to have it now, just a send pull request.

n8v8R, 2020-10-10:
> And now you trying to offend me.

Hmm? You are cited as maintainer of the package and it has been politely asked to update the package to be on par with its source development. 
As reaction you are questioning the motivation of keeping the package on par with its source and asking for a PR.

Either you want to maintain the package or you do not, and if latter that is fine, simply say so, instead of starting such silly discourse and being accusatory that a user aims to offend you.
____
> Unfortunately, this is not how it works. If you want to have it now, just a send pull request.

Apparently not with you as package maintainer, others take a look at the source development and decide whether or not to upgrade the package in the OpenWrt repository.

BKPepe, 2020-10-10:
Closing. This is not going anywhere. You should use your time more precisely or give efforts somewhere else. E.g. pull request.

closed 2020-10-10 by BKPepe

diizzyy, 2020-10-10:
1.25.0 doesn't really offer much "enhancement" over the current version as far as OpenWrt is concerned (many metrics are stripped out of the kernel anyway by default). We both are doing work on voluntary basis and have other interests than OpenWrt not to mention that **testing** packages for OpenWrt is very tedius compared other distributions due to what kind of devices OpenWrt targets. We could blindly update stuff just because it compiles but you could potentially end up with a broken application instead (has occurred on multiple times in tree) which I'd say would be more harmful and it's been an ongoing issue for years now in general regarding this repo. If you want more frequent updates I'd suggest that you'd look at another distro such as Alpine, Arch, Debian using the testing branch etc or as Josef suggested do a pull request (testing required).

n8v8R, 2020-10-10:
> 1.25.0 doesn't really offer much "enhancement" over the current version as far as OpenWrt is concerned (many metrics are stripped out of the kernel anyway by default).

There are users compiling their own kernel, e.g. my node currently runs fine on kernel 5.9-rc8, with metrics back in place, but that is beside the point.
____
> We both are doing work on voluntary basis and have other interests than OpenWrt

That is naturally understood and not been questioned, as well as understanding the package being none-essential for the node's functionality.
____

> **testing** packages for OpenWrt is very tedius compared other distributions due to what kind of devices OpenWrt targets. We could blindly update stuff just because it compiles but you could [...]

That cannot be helped by the user and is supposedly part of the maintenance effort for any sort of package and any frequency of package releases.
____
> or as Josef suggested do a pull request (testing required).

in such case it would be unlikely for the user open an issue, such as this in the package feed, or else the user would have created a PR in the first place.

It should be rather implied for the stated maintainer that user is not likely to come up with a PR and testing. Therefore for the maintainer going into a discourse about the motivation of updating the package and asking for a PR is entirely moot, instead it would be expected of the maintainer to:

* update the package and close the issue in the package feed
* not update the package, preferably with a reasoning, and close the issue in the package feed

OdysLam, 2020-10-12:
Hi, developer relations from Netdata here. Happy to meet you all :) 

We will be rolling out a pretty amazing new feature these next days, which will require Netdata version 1.26 that we will roll out  in the week. We also have a large number of bug fixes and improvements, so I am sure that we wouldn't want to left out our OpenWRT users.

Is there anything we can do from our part to help you maintain the package list more often/with less friction? 

Best,
Odysseas

BKPepe, 2020-10-12:
Hi @OdysLam,

No, you don't need to worry about that. Nothing wrong to you or anyone. Just check @n8v8R's issues and so on. I have such experiences on other social media networks.
He needs to understand that he can not whistle (hopefully this is used in English as well) or want something and everyone will do that ASAP. Everyone has its own priorities, job, and such.

We are regularly updating Netdata, we just didn't update it because of the issue, which I met in 1.24.0 and I could backport patch from 1.25.0 or wait for until a new release. However, I didn't get to it since then.

OdysLam, 2020-10-12:
That's good to hear @BKPepe. We want to facilitate the people who make our software available to your communities. You are humbled for your volunteering work. 

Please, do join our [community](https://community.netdata.cloud), so you can easily reach me out with a new post if you have any questions or you need any assistance in facilitating your efforts on maintaining the package. 

I will circle back here when we release our new version, so you can test it and hopefully merge it into `main` :) .

Thanks again and cheers !

diizzyy, 2020-10-12:
I'll only echo what Josef said and frankly I didn't see the need in this case. A downside as far as OpenWrt is concerned is that each version grows quite significantly which requires more work slimming it down given the target devices. We probably should at some point (tm) make separate packages for scripts but that requires more work etc to maintain....