openwrt/packages #13646: 6rd: IPv6 is still configured when ipv6 is disabled on wan interface

joelvdvoort, 2020-10-12:
Maintainer: @dedeckeh 
Environment: mvebu-cortexa9, Turris Omnia, Turris OS 5.1.2 (based on OpenWrt 19.07.4), package 6rd 9-4

Description:

IPv6 is configured by 6rd package when ipv6 is disabled on the wan interface.

```
config interface 'wan'
	option ifname 'eth2'
	option proto 'dhcp'
	option ipv6 '0'
```

When running `ip a` the 6rd_wan interface is there:

```
156: 6rd-wan_6@eth2: <NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit [redacted_ip4] brd 0.0.0.0
    inet6 [redacted_ip6] scope global 
       valid_lft forever preferred_lft forever
    inet6 [redacted_ip6] scope global 
       valid_lft forever preferred_lft forever
```
I'm not sure if this is a bug or intended (albeit opaque) behavior. I'd expect 6rd to respect the options set on the wan interface. If there's anything I can do to help. Please let me know!

dedeckeh, 2020-10-12:
The config option ipv6 on interface level is only relevant for PPP interfaces as it will instruct the ppp daemon not to negotiate IPv6 settings like the link local address.
It will not be used as any indication for an (IPv6) interface stacked on top of this interface not to come up

closed 2020-10-12 by dedeckeh

joelvdvoort, 2020-10-12:
I'm sorry. I posted the wrong snippet from my configuration. The relevant part should be: 

```
config interface 'wan6'
option ifname '@wan'
option proto 'none'
```