openwrt/packages #13652: Boost: can't be linked with stdcpp automatically

brvphoenix, 2020-10-13:
Maintainer: @ClaymorePT @neheb 
Environment: x86_64

Description:

Find an issue that the package stdcpp in the `DEPENDS` can't be linked automatically. 
Check the linked libs with `ldd`, and get the errors as following:
```
/usr/lib/libboost_chrono.so.1.74.0
	ldd (0x7fd4a5872000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fd4a5854000)
	libc.so => ldd (0x7fd4a5872000)
Error relocating /usr/lib/libboost_chrono.so.1.74.0: _ZTVN10__cxxabiv117__class_type_infoE: symbol not found
Error relocating /usr/lib/libboost_chrono.so.1.74.0: _ZTVN10__cxxabiv117__class_type_infoE: symbol not found
Error relocating /usr/lib/libboost_chrono.so.1.74.0: _ZTVN10__cxxabiv117__class_type_infoE: symbol not found
Error relocating /usr/lib/libboost_chrono.so.1.74.0: _ZTVN10__cxxabiv120__si_class_type_infoE: symbol not found
Error relocating /usr/lib/libboost_chrono.so.1.74.0: _ZTVN10__cxxabiv120__si_class_type_infoE: symbol not found
Error relocating /usr/lib/libboost_chrono.so.1.74.0: _ZTVN10__cxxabiv120__si_class_type_infoE: symbol not found
Error relocating /usr/lib/libboost_chrono.so.1.74.0: _ZTVN10__cxxabiv120__si_class_type_infoE: symbol not found
```
If add the `-lstdc++` to `TARGET_LDFLAGS`, everything is OK. 
```
ldd /usr/lib/libboost_chrono.so.1.74.0
        ldd (0x7ff624929000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7ff6247e6000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7ff6247d2000)
        libc.so => ldd (0x7ff624929000)
```

However, if some packages depend on boost libs and are linked directly or indirectly with `libstdc++.so.*`, it seems that they can also woks well. Therefore, is it necessary to link the boost library by adding `-lstdc++` to `TARGET_LDFLAGS` manually.

Test the boost on archlinux, and it seems that the libstdc++ is linked dynamically.
```
ldd /tmp/libboost_chrono.so.1.72.0
        ldd (0x7f056eefc000)
        librt.so.1 => ldd (0x7f056eefc000)
        libpthread.so.0 => ldd (0x7f056eefc000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7f056edb6000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f056eda2000)
        libc.so.6 => ldd (0x7f056eefc000)
```

neheb, 2020-10-13:
Am I reading right that you’re running ldd on host arch Linux libraries?

ClaymorePT, 2020-10-13:
I'm wondering if this is a Boost Makefile issue or a global OpenWRT build manager issue. If the dependency is specified in the `DEPENDS` field, then it should be linked. I wonder if the other libs specified in that field are also having the same issue.

brvphoenix, 2020-10-13:
> I wonder if the other libs specified in that field are also having the same issue.

Not sure. List two different packages that depend on libstdcpp.

libjsoncpp
```
ldd /tmp/libjsoncpp.so.24
        ldd (0x7f48a4ca0000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7f48a4b3f000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f48a4b2b000)
        libc.so => ldd (0x7f48a4ca0000)
```
libpcre2
```
ldd /tmp/libpcre2-8.so.0.10.0
        ldd (0x7f3d205da000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f3d20551000)
        libc.so => ldd (0x7f3d205da000)
```

neheb, 2020-10-13:
@ClaymorePT eh no. Linking is done by each package's Makefile. The DEPENDS fields is mostly to add a package as a build dependency. There's also logic that checks if a dependency is missing.

ClaymorePT, 2020-10-13:
@neheb So basically this has been going on forever and we are only noticing now thanks to @brvphoenix inspection? Auch...

Okay @brvphoenix  I understand what you are saying, lets add it to `TARGET_LDFLAGS` since it should have already been there.

@brvphoenix I can only work on this issue later in day. If you have a fix ready to submit, feel free to do push it.

neheb, 2020-10-13:
If that were the case. several packages would be broken.

It's possible this regressed in some recent version.

PR #13676 (2020-10-19): boost: improve package compilation
Maintainer: @ClaymorePT @neheb
Environment: x86_64

Description:
* Add `-lstdc++` to `LDFLAGS`, and close #13652
* Add `-Wl,--gc-sections,--as-needed` to  `LDFLAGS` and `-ffunction-sections -fdata-sections -flto` to `CFLAGS` which will reduce the size by 10% ~ 13%
* Only a virtual package will be created if only static libs are built, which will avoid compiliation failures
* Add python3-base to boost-python3's dependences according to [python.jam](https://github.com/boostorg/build/blob/develop/src/tools/python.jam#L1010~#L1014)
closed 2020-10-19 by neheb