openwrt/packages #13674: golang: option GOLANG_MOD_CACHE_WORLD_READABLE not working

sch-m, 2020-10-15:
Maintainer: @jefferyto 
Environment: x86_64, master

Description:
Commit 3dd55b504cb4 moved the module cache into DL_DIR.
For backup reasons, I need the content of my DL_DIR world-readable, so I've enabled the new option GOLANG_MOD_CACHE_WORLD_READABLE, but this does not work for all directories/files:

```
find . -type d -not -perm -go+rx:

./golang.org/x/sys@v0.0.0-20190422165155-953cdadca894
./gopkg.in/yaml.v2@v2.2.2
./gopkg.in/alecthomas/kingpin.v2@v2.2.6
./github.com/beorn7/perks@v1.0.0
./github.com/matttproud/golang_protobuf_extensions@v1.0.1
./github.com/hashicorp/golang-lru@v0.5.1
./github.com/go-logfmt/logfmt@v0.4.0
./github.com/pkg/errors@v0.8.1
./github.com/golang/protobuf@v1.3.2
./github.com/tarm/goserial@v0.0.0-20151007205400-b3440c3c6355
./github.com/go-kit/kit@v0.9.0
./github.com/prometheus/common@v0.7.0
./github.com/prometheus/procfs@v0.0.2
./github.com/prometheus/client_golang@v1.0.0
./github.com/prometheus/client_model@v0.0.0-20190129233127-fd36f4220a90
./github.com/sirupsen/logrus@v1.4.2
./github.com/alecthomas/template@v0.0.0-20190718012654-fb15b899a751
./github.com/alecthomas/units@v0.0.0-20190717042225-c3de453c63f4
./github.com/xlab/at@v0.0.0-20190216203222-7884b63edf83
```

jefferyto, 2020-10-16:
Thanks - the permissions in an existing module cache directory should be fixed by the next Go package build after #13681 is merged.

sch-m, 2020-10-16:
Nice. Thank you.

closed 2020-10-16 by neheb