openwrt/packages #13696: Request for a `python3-pynacl` package

ivankravets, 2020-10-17:
- https://pypi.org/project/PyNaCl/

Thanks!

BKPepe, 2020-10-17:
This package is available in master branch and it will be part of upcoming OpenWrt 20.xx release.
See: https://github.com/openwrt/packages/blob/master/lang/python/python-pynacl/Makefile

If you want to have it in OpenWrt 19.07, you will need to compile it yourself.

closed 2020-10-17 by BKPepe

ivankravets, 2020-10-17:
How to make it available in Entware via `opkg`? I can’t compile it directly on my machine, it depends in libffi-dev.

BKPepe, 2020-10-17:
In this case, I suggest reaching Entware guys. I am not familiar with how they are doing that. I am sorry.

ivankravets, 2020-10-17:
Sorry, it seems I found correct place for this issue. https://github.com/Entware/Entware/issues/549