openwrt/packages #13708: Package/frr-vtysh DEPENDS ‘more’=‘moreutils’ ?

guyezi, 2020-10-19:
#The 'define package / FRR vtish' in 'packages / FRR / makefile' depends on 'more'. Is this' more 'just' moreutils'?

I'm sorry, my English is not good, this "issues" is done by translation software. Hope no translation error!

Maintainer: @\Lucian Cristian <lucian.cristian@gmail.com>
Environment: (master)

Description:

```
define Package/frr-vtysh
  $(call Package/frr/Default)
  DEPENDS+=+frr-libfrr +libreadline +libncurses **+more**
  TITLE:=integrated shell for frr routing software
  CONFLICTS:=quagga-vtysh
endef 
```
The following problems occurred during feeds install:：

```
WARNING: Makefile 'package/feeds/packages/frr/Makefile' has a dependency on 'more', which does not exist
```

BKPepe, 2020-10-19:
ping @lucize

lucize, 2020-10-19:
please update your sdk or just import more package from base repository, it was added some time ago

[image]

guyezi, 2020-10-20:
@lucize 

I used 19.07.4 source code compilation, not the image of the SDK. There is no "more" in 19.07.4 source code.

I looked through all the ‘Makefile’ in Packages and found no “more”.

[image]

lucize, 2020-10-20:
> please update your sdk

look into master