openwrt/packages #13712: Eror sdk rpi4 snapshoot

predators46, 2020-10-19:
`make[2]: Leaving directory '/home/azurevps46/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/feeds/base/package/firmware/prism54-firmware'
time: package/feeds/base/prism54-firmware/compile#0.95#0.51#6.58
package/Makefile:112: recipe for target 'package/kernel/linux/compile' failed
make[1]: [package/kernel/linux/compile] Error 2 (ignored)
make[2]: Entering directory '/home/azurevps46/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64'
make[2]: *** package/kernel/linux: No such file or directory.  Stop.
make[2]: Leaving directory '/home/azurevps46/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64'
time: package/kernel/linux/compile#0.00#0.00#0.00
package/Makefile:112: recipe for target 'package/kernel/linux/compile' failed
make[1]: *** [package/kernel/linux/compile] Error 2
make[1]: Leaving directory '/home/azurevps46/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64'
/home/azurevps46/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64/include/toplevel.mk:227: recipe for target 'package/luci-app-ssr-plus/compile' failed
make: *** [package/luci-app-ssr-plus/compile] Error 2
azurevps46@openwrt:~/openwrt-sdk-bcm27xx-bcm2711_gcc-8.4.0_musl.Linux-x86_64$    `

BKPepe, 2020-10-19:
This issue does not belongs to packages repository. You should report it to OpenWrt, but it should be fixed in upcoming days.
Related to https://github.com/openwrt/openwrt/pull/3520

closed 2020-10-19 by BKPepe

predators46, 2020-10-19:
> This issue does not belongs to packages repository. You should report it to OpenWrt, but it should be fixed in upcoming days.
> Related to [openwrt/openwrt#3520](https://github.com/openwrt/openwrt/pull/3520)

Thanks