openwrt/packages #13726: stubby: Stubby spams log with “Could not schedule query” when internet goes down

rbubley, 2020-10-20:
Maintainer: @jonathanunderwood
Environment: BT Home Hub 5A, xRX200 rev 1.2, OpenWrt 19.07.4

Description:

When the WAN goes down (which I have been unlucky to have happen a lot recently due to local problems), stubby fills the log with this message (about 110 a second):
```
Tue Oct  6 12:22:41 2020 daemon.err stubby[4098]: Could not schedule query: None of the configured upstreams could be used to send queries on the specified transports
```

I have (I believe) a standard stubby configuration, having followed the guide:
```
stubby.global=stubby
stubby.global.manual='0'
stubby.global.trigger='wan'
stubby.global.dns_transport='GETDNS_TRANSPORT_TLS'
stubby.global.tls_authentication='1'
stubby.global.tls_query_padding_blocksize='128'
stubby.global.appdata_dir='/var/lib/stubby'
stubby.global.edns_client_subnet_private='1'
stubby.global.idle_timeout='10000'
stubby.global.round_robin_upstreams='1'
stubby.global.tls_connection_retries='2'
stubby.global.listen_address='127.0.0.1@5453'
stubby.@resolver[0]=resolver
stubby.@resolver[0].address='1.1.1.1'
stubby.@resolver[0].tls_auth_name='cloudflare-dns.com'
stubby.@resolver[1]=resolver
stubby.@resolver[1].address='1.0.0.1'
stubby.@resolver[1].tls_auth_name='cloudflare-dns.com'
stubby.@resolver[2]=resolver
stubby.@resolver[2].address='8.8.4.4'
stubby.@resolver[2].tls_auth_name='dns.google'
stubby.@resolver[3]=resolver
stubby.@resolver[3].address='8.8.8.8'
stubby.@resolver[3].tls_auth_name='dns.google'
```
I would have hoped that there would be some sort of backoff after a failure. As it is, it is quite difficult to see what has happened as the logs get completely filled with these messages.

rbubley, 2021-01-20:
Ok, a little more info.

For example, when the router boots, stubby issues these logs:

```
2021-01-19T14:20:10+00:00 OpenWrt stubby[4055]: [14:20:10.754019] STUBBY: Stubby version: Stubby 0.3.0
2021-01-19T14:20:11+00:00 OpenWrt stubby[4055]: [14:20:11.652328] STUBBY: Read config from file /var/etc/stubby/stubby.yml
2021-01-19T14:20:11+00:00 OpenWrt stubby[4055]: [14:20:11.705589] STUBBY: DNSSEC Validation is OFF
2021-01-19T14:20:11+00:00 OpenWrt stubby[4055]: [14:20:11.705768] STUBBY: Transport list is:
2021-01-19T14:20:11+00:00 OpenWrt stubby[4055]: [14:20:11.705868] STUBBY:   - TLS
2021-01-19T14:20:11+00:00 OpenWrt stubby[4055]: [14:20:11.705976] STUBBY: Privacy Usage Profile is Strict (Authentication required)
2021-01-19T14:20:11+00:00 OpenWrt stubby[4055]: [14:20:11.706081] STUBBY: (NOTE a Strict Profile only applies when TLS is the ONLY transport!!)
2021-01-19T14:20:11+00:00 OpenWrt stubby[4055]: [14:20:11.711973] STUBBY: Starting DAEMON....
2021-01-19T14:20:13+00:00 OpenWrt stubby[4055]: Could not schedule query: None of the configured upstreams could be used to send queries on the specified transports
2021-01-19T14:20:13+00:00 OpenWrt stubby[4055]: Could not schedule query: None of the configured upstreams could be used to send queries on the specified transports
2021-01-19T14:20:13+00:00 OpenWrt stubby[4055]: Could not schedule query: None of the configured upstreams could be used to send queries on the specified transports
2021-01-19T14:20:13+00:00 OpenWrt stubby[4055]: Could not schedule query: None of the configured upstreams could be used to send queries on the specified transports
```

The last line repeats roughly 12 times per second, and then at some point post boot they just stop.

jonathanunderwood, 2021-01-21:
Yes, I see the same thing. This is probably better reported upstream though.

dhewg, 2021-10-13:
Same issue here, If the internet connection is lost this spam fills up `logread` in a manner of seconds.

stubby can already be configured to be started via procd interface triggers. It could just do the inverse on top, stop it if the interface is down.

/etc/init.d/stubby:
```
    for trigger_item in $trigger                                                                                                                                                                             
    do                                                                                                                                                                                                       
        procd_add_interface_trigger "interface.*.up" "$trigger_item" "$stubby_init" start                                                                                                                    
        procd_add_interface_trigger "interface.*.down" "$trigger_item" "$stubby_init" stop                                                                                                                   
    done               
```

Seems to work for me

rbubley, 2021-11-17:
@jamesmacwhite would the suggestion above to take stubby down work for you in an mwan setting, or would it take down stubby if _any_ of your interfaces went down?

jamesmacwhite, 2021-11-17:
@rbubley I don't use stubby anymore as I found it a bit unreliable with mwan3 and DNSSEC issues with some sites.

My workaround was to restart stubby on an ifup event in the /etc/mwan3.user file, but modifications to the init.d script is the better way as above I'd imagine.

dhewg, 2021-11-17:
FWIW I switched from the above suggestion to this patch, maybe it's useful for you guys:
[0001-stubby-reduce-epic-log-spam-when-offline.patch.txt](https://github.com/openwrt/packages/files/7552998/0001-stubby-reduce-epic-log-spam-when-offline.patch.txt)